ControlSystemStudio/cs-studio #871: Option to crop rather than zoom ill-fitting OPIs in LinkingContainers

willrogers, 2015-01-08:
If you put an OPI in a LinkingContainer (and it doesn't fit), you can choose either 

* zoom to fit the OPI in
* scrollbars

For small OPIs, scrollbars are unlikely to be correct, and zoom tends not to fill the available space.  EDM has two options:

* scrollbars
* crop OPI to size of LinkingContainer

Could we add a third option to BOY, so that the options are 

* zoom to fit the OPI in
* scrollbars
* crop OPI to size of LinkingContainer

?  It's fairly easy to do; the problem would be in choosing the XML format and backward compatibility.

kasemir, 2015-01-08:
If I understand correctly, this is currently based on the bool options `auto_size` and `zoom_to_fit`.
The scroll bars appear automatically if both are false, and the linked *.opi is larger than the container.

Wouldn't the complete set of options be this?
 * Size *.opi to fit the container
 * Size the container to fit the linked *.opi
 * Don't resize anything, crop if *.opi to large for container
 * Don't resize anything, add scrollbars if *.opi to large for container

I think it'd be best to find that list of options, then use a new `resize_behavior` option with those (4?) values. As the widget loads from a previous version, it would need to handle it:
```
if (old_version)
{
   if (zoom_to_fit)
      resize_behavior = fit_container;
   else 
     resize_behavior = scroll;
}
```

willrogers, 2015-01-09:
Re: your second option.  Is a widget allowed to display larger than its specified bounds?

kasemir, 2015-01-09:
Re: Size the container to fit the linked *.opi:
No, a widget is not allowed to display larger than its specified bounds.
Currently, the auto_size is really an edit-time option. The widget changes its size to match the linked *.opi.
But a widget can also change its size and position at runtime, some scripts do that. So while the widget should not draw outside its current size, it could change its current size to match the linked *.opi, and then the drawing is all within bounds.

closed 2015-01-16 by berryma4