ControlSystemStudio/cs-studio #880: Memory Leak in BOY

jbobnar, 2015-01-14:
There is a serious memory leak in BOY. Any OPI that is opened is retained in memory forever. An easy way to see this is to open and close a high-memory-consuming opi (...opibuilder.imagelib/demo/PID_Water_Cooling.opi is a good example). If you repeat the open/close many times, you can see the memory consumption going up. Using a profiler, one can see that the OPIRunner and other opi elements are being retained in memory.

There are 2 problems:
1. In the whole chain of classes and delegates (like OPIRunner and OPIRuntimeDelegate) some of the things are never disposed of. 
2. Every contribution to the toolbar is retained in the ToolBarManagerRenderer until shutdown.

The problem 1. is fairly straightforward to fix and already fixes 99% of the leak. One just need to make sure that the dispose calls are properly forwarded.
The problem 2. is more difficult. Every time when an OPI is opened the OPIRunnerToolbarContributor regenerates the toolbar items, the items are added to the ToolBarManagerRenderer (modelToContribution) and are never removed. I think the definition of the toolbar items should be done in the plugin.xml (through extensions), similar to how it is done in the scantree case.

kasemir, 2015-01-15:
1. In the whole chain of classes and delegates (like OPIRunner and OPIRuntimeDelegate) some of the things are never disposed of. 
--
I wonder if this also happens with the code from the boy_runtime_view branch.

jbobnar, 2015-01-15:
There is no difference with boy_runtime_view branch. Drilling deeper it appears that the root cause is in eclipse. The bug is similar to this one: https://bugs.eclipse.org/bugs/show_bug.cgi?id=206584

After opening a new WorkbenchWindow the instance is referenced from all over the place and never released. No solution found yet.

jbobnar, 2015-01-19:
I found at least 3 bugs related to this issue
https://bugs.eclipse.org/bugs/show_bug.cgi?id=206584
https://bugs.eclipse.org/bugs/show_bug.cgi?id=436225
https://bugs.eclipse.org/bugs/show_bug.cgi?id=354538
The problem can be easily reproduced with any version of eclipse of 3.4 (perhaps even earlier) and later. 

It appears that we will have to wait at least until eclipse 4.5.

The commit be62294 disposes of the OPI related objects, which brings the memory leak to ~10-100 kB for every new workbench window that is opened and closed.

closed 2015-02-19 by jbobnar