ControlSystemStudio/cs-studio #884: Eclipse contributed menu items are not relevant in CS-Studio

mfurseman, 2015-01-16:
There menu items included with some of the Eclipse plugins, however they don't seem suitable in CS-Studio. For instance the 'File->Convert Line Delimiters To' menu item, and the ones outlined below to perform file navigation in an Editor:

[image]

Although there may be applications for these items they are distracting when using Diamonds implementaion of CSS.

The menu actions can be removed using an internal API:
https://github.com/dls-controls/cs-studio/compare/ControlSystemStudio:4.0.x...menu-cleanup-4.0.x

berryma4, 2015-01-16:
> dls-controls@ControlSystemStudio:4.0.x...menu-cleanup-4.0.x

I agree with this change, can you submit a pull request. @kasemir , your
thoughts?

kasemir, 2015-01-16:
Yes, good!
Just add comments to the code that it's intenal API, and why we use it.

kasemir, 2015-01-21:
Somewhat related, in case somebody later checks this ticket about unwanted menu entries:
`NoResourceEditorInput` from https://github.com/ControlSystemStudio/cs-studio/blob/master/core/plugins/org.csstudio.ui.util/src/org/csstudio/ui/util/NoResourceEditorInput.java can be used to remove irrelevant context menu entries for "Editors" that happen to related to a "IFile".

kasemir, 2015-02-02:
I think this can be closed, since the related pull requests have been merged.

closed 2015-02-02 by kasemir

berryma4, 2015-02-02:
Sorry, I usually close the issue after it gets merged to master.