ControlSystemStudio/cs-studio #899: remove utility.pv

kasemir, 2015-01-21:
alarm server, archive engine, scan server, PV table, EPICS PV Tree use vtype.pv.
opibuilder, probe use (via simplepv) either utility.pv or PVManager.

Remove utility.pv; have simplepv select between vtype.pv or PVManager.

kasemir, 2015-01-26:
On the bug899_util_pv branch,  utility.pv dependencies have been removed from the opibuilder (via simplepv).

The older o.c.diag.probe plugin has also been updated to use vtype.pv instead of util.pv, but otherwise left as is since products tend to include o.c.diag.probe.pvmanager anyway.

The only code that still depends on the utility.pv.* plugins is now `org.csstudio.archive.reader.archiverecord`. It doesn't seem worth updating, however, since it doesn't return any data. Its `ArchiveRecordReader` class is where Jan left it 2 years ago with a TODO to return VType values.

@berryma4 : Exclude archiverecord from pom.xml so it's not built, then remove all  utility.pv.* plugins?

berryma4, 2015-01-26:
That sounds fine with me.
Thank you!