ControlSystemStudio/cs-studio #905: Scan Server PVA Server

kasemir, 2015-01-23:
The Scan Server PVA Server was implemented based on org.epics.pvaccess and org.epics.pvdata.

These have been replaced with org.epics.pvAccessJava and org.epics.pvDataJava.

pvmanager and vtype.pv have been updated, and the two versions of the V4 protocol cannot coexist on the network, so the scan server PVA server needs to be updated or disabled until it can be updated.

@berryma4 : Are you using these PVA PVs?

kasemir, 2015-01-23:
Simply replacing the plugins results in  24 errors because ServerContextImpl is now internal API, and pvData API has changed.

kasemir, 2015-01-23:
Also, doesn't show as compile error but "uri:ev4:nt/2012/pwd:NTTable" now needs to be "epics:nt/NTTable:1.0".

berryma4, 2015-01-23:
I'm not using it anymore, ... although I wish I could in 2 weeks (up coming experiment).
I think the whole thing needs to be rethought.  I know @shroffk was interested :8ball:

kasemir, 2015-01-23:
We now have V4 servers and clients on an operational beamline, using the current release V4.4.
If you try to access current pvAccess servers with older pvAccess clients, they hang up.
If I try to access the scan server with its older pvAccess with a new client, I get nothing.
So you need an older copy of both scan server and CSS to use this right now.

I'd like to disable the PVA server code in the scan server ASAP, since older V4 code on our production beam lines worries me and I don't have time to update it in the next few weeks..
I'm for eventually putting it back in, based on basically starting over, beginning by reconsidering the channel name handling based on the URL agreement from the last developers meeting.

kasemir, 2015-01-23:
PVA URL notes from the meeting can be found at the end of https://docs.google.com/document/d/1iKe4iOwfbIPs7foBOhVSsJWzeNNESsVIaAPie4Sc5UA/pub

kasemir, 2015-02-04:
Progressing on #911 will revert the pvdata and pvaccess plugin names back from pvDataJava and pvAccessJava, but the API is still the current one, not supported by the PVA server code in the scan server.
As #911 resolves, I'll disable the PVA server in the scan server.

glcj, 2016-03-21:
Hello,
As it forms the URL to access the fields of a structure such as that shown.
I have tried various combinations in CSS-Probe, but none has worked.
**pva://Fuente**
Thank you for your help,

8     | Fuente         |
      |                | record Fuente epics:nt/NTScalar:1.0
      |                |     int value 100
      |                |     valueAlarm_t valueAlarm
      |                |         boolean active false
      |                |         int lowAlarmLimit 0
      |                |         int lowWarningLimit 0
      |                |         int highWarningLimit 0
      |                |         int highAlarmLimit 0
      |                |         int lowAlarmSeverity 0
      |                |         int lowWarningSeverity 0
      |                |         int highWarningSeverity 0
      |                |         int highAlarmSeverity 0
      |                |         int hysteresis 0

berryma4, 2016-03-24:
PVA Scanserver is no longer implemented.  It might be easier, better, to make a scanserver datasource for pvmanager (using its rest api), and a pvmanager service to submit scans.