ControlSystemStudio/cs-studio #912: BOY: Connector between linking containers is not correctly loaded

ayemos, 2015-02-04:
There's a problem with loading/saving Connection Models from/to opi file.

The problem can be reproduced by trying to create Connection between a widget inside LinkingContainer and a widget outside, and to {save it and load it, go to runtime mode, and so on...}.

The issue is about the way BOY loads LinkingContainer from opi file which first tries to just create LinkingContainer itself and load linked opi file after all other widget(includes connection models) are loaded.
In this way, some connections which have one or two endpoints inside linking container can't find those widgets on endpoints when it is created so it can't be initialized and aren't shown on editor and/or runtime mode.

As an addition I made changed to fix the problem on trying to load LinkingContainer which has load-loop in .opi file hierarchy. Before, it just looked for a violation of loading itself(load A.opi from a LinkingContainer on A.opi) so it wasn't able to detect cycle: A.opi loads B.opi which loads C.opi which loads A.opi...

kasemir, 2015-02-12:
Did the merge resolve the issue?

kasemir, 2015-05-06:
Too late now, but for the record this was a bad idea.

The content of a linking container is not meant to be known outside of the linking container.
The parent *.opi cannot make any assumptions about widgets inside a linking container, because the content of the linked file can change. Only the macros required by the linked file will hopefully stay constant as the exact content of the linked *.opi is modified.
In addition, many users want to select a different file for a linking container at runtime. 

--> The widgets inside a linking container must remain private to the linking container. Connectors from the parent *.opi to widgets inside the linking container are conceptually not possible/permitted.

Unfortunately, the original implementation of the widget model merged the content of linking containers into one big model, so access to widgets inside the linking container _was_ possible. This update used the flaw in the original implementation to permit connectors to widgets inside linking containers. The exact implementation added `if (widget instance of LinkingContainer)` type checks which hint to a class hierarchy design problem.

kasemir, 2015-07-06:
We should close this as "won't fix".

The content of the linking container is private to the linking container. When a linking container displays "embedded.opi", the content of "embedded.opi" can be edited over time. Or the linking container is used to switch between "one_embedded.opi" and "another_embedded.opi". In any case, the display that contains the linking container cannot make any assumptions regarding the content of the linking container.

The current implementation of the BOY editor is somewhat cute in displaying the content and even allowing you to select widgets inside the container and edit them, but that's misleading because such changes will _not_ be saved to the "embedded.opi".

The current BOY runtime does merge the embedded content into one big display model and such allow access to the embedded displays, but that is a side effect of the current implementation, it is not a decision for the behavior of the linking container.

So in summary, connections to widgets inside a linking container are not meant to be supported.

closed 2015-07-06 by kasemir

utzeln, 2016-02-25:
Just a comment - if I understand the statement "The widgets inside a linking container must remain private to the linking container. Connectors from the parent *.opi to widgets inside the linking container are conceptually not possible/permitted", on ITER we are using the linking container as a way to manage HMI object library.

Our use case is that we define a component such as a control valve by specifying an EPICS template, an HMI object representation and a faceplate.  For instance a control valve in BOY editor looks like that:
[image]
And clicking on such object, will open automatically its faceplate.

The library then includes the HMI objects as an OPI files, as well as the faceplates. This way the user of the library can insert a control valve in his synoptic by creating a linking container to the HMI object OPI file and instantiate the macros.

And of course, connectors between control valves are required - i.e. connectors between widgets in 2 different linking containers.

berryma4, 2016-02-25:
Could you add your screen to the tests, so we can assure compatibility:
https://github.com/ControlSystemStudio/opi-regression-tests
You can put it in the Issues directory.

kasemir, 2016-02-25:
> connectors between control valves are required - i.e. connectors between widgets in 2 different linking containers.

The inside of the linking container can't be known. The content is in a separate file, the content can change, so how can you make a connection to a widget inside a linking container?
BOY allows that because of a somewhat broken implementation: It loads the current content of the linked file, allows connections to the widgets in there, but those widgets can change the next time you open the file.

What might be possible is a connection to the linking container itself, to some point on the border of the linking container.

berryma4, 2016-02-25:
I'm sorry if I misunderstood

> What might be possible is a connection to the linking container itself, to some point on the border of the linking container.

Is exactly what I was assuming here.

kasemir, 2016-02-25:
In addition to tying connections to a point on the border of a LinkingContainer it might be possible to tie a connection to some arbitrary location inside the container, i.e. define the endpoint of the connection with some offset from the origin of that widget.
That way, the connection would start or end somewhere in the area of the linking container.
If the linking container happens to contain a widget right there it looks like you're connected to that widget, but the connection really goes to the container, i.e. a widget in the same display file, not to an embedded widget, since what's actually embedded in the container isn't stable.

utzeln, 2016-02-25:
I did not want to reopen the discussion thread as I understand the concern - we will have to fix this problem for ITER site - knowing the limitations you pointed out.
What I wanted to clarify is the need of having an "HMI object library" to be reused easily by users. They are not widgets that can be dragged and dropped from the palette, they combine multiple widgets/PVs and can be instantiated using macros.
I do not know if it is part of the specifications of the new tools under development.

kasemir, 2016-02-25:
OK, understand.

For such a library, using embedded widgets is a good idea exactly because a 'gadget.opi', embedded in all the screens that need to display a gadget, allows you to change the 'gadget.opi' and all the screens that use it will get the updated version.

I think the only new feature you'd need to better support this would be a read-only editor. That way you create an "HMI object library.opi" file that has linking containers with all those gadgets. It's opened in the editor together with the file that you're actually editing. To add an object, you drag it from the "HMI object library.opi" that's open in one editor panel and drop it into the other editor panel that you're actually editing.
You can already do that, except you need to be careful to press "Control" to get a copy-drop, instead of removing the gadget from your "HMI object library.opi". With a read-only editor, there's be no need to remember pressing "Control". The implementation of read-only editor could simply look at the file permissions: If it can't "save" the file because it's not writable, then don't allow modifications in the editor.

With BOY, you can get performance problems when using many embedded displays because it loads all the embedded screens, one by one, on the UI thread, so CSS freezes until it's all done.
With the display builder update, nothing freezes. As soon as the main *.opi is loaded, it shows. All the embedded displays are then loaded in parallel background threads, so they appear whenever they're ready. When I use that with our instrument displays it's overall faster than BOY even without having implemented 'caching' of *.opi files, i.e. simply loading the same "motor.opi" N times into a motor overview.