ControlSystemStudio/cs-studio #976: DataBrowser reload archives

tashasummers, 2015-02-23:
Could databrowser periodically reload archived data? With the default live buffer of 5000 a space grows between the data loaded from archive when the plot is first launched and the maximum live buffer size until the archived data falls off the end. Is the only way around this to have buffer sizes of 30K or 90K for 8 or 12 hours plots? I use these scales frequently for status displays.

kasemir, 2015-02-23:
Check the option to "Automatically refresh historic data when live sample buffer is full" in the data browser preferences.

kasemir, 2015-02-23:
"Is the only way around this to have buffer sizes of 30K or 90K for 8 or 12 hours"..
If you want to see 'life' data for that time range: yes.
The automated history refresh will fill the gap, that space that otherwise grows between the historic and the live data with samples fetched from the archive, which tends to have a coarser resolution.

kasemir, 2015-02-23:
Pardon my English.  life -> live. Data obtained from the running PV, not from the archive.

tashasummers, 2015-02-23:
I see this in version 4, thank you, it is a nice addition.

kasemir, 2015-03-18:
I'm closing this because it's available in version 4, and there's no plan to implement this in the now quite different 3.x code

closed 2015-03-18 by kasemir