ControlSystemStudio/cs-studio #977: BOY changed how it resolves paths to displays opened in actions

kasemir, 2015-02-23:
We have displays created with 3.x that are linked as follows:

main.opi has linking container for ../some/place/other.opi.
That other.opi than has an action button for ./final.opi, located in that same ../some/place, i.e. in the same directory as other.opi.

Used to work fine, because the relative path to the action was resolved based on the location of where it's defined, i.e. other.opi.
Now it's resolved based on when it's called, i.e. relative to main.opi, and then fails because there is no final.opi in the directory of main.opi.

kasemir, 2015-02-25:
The `boy_runtime_view` branch does not exhibit this problem, so it's not a 3.x vs. 4.x issue.
Have slight suspicion that #913 introduced this behaviour less than a month ago, since it changed the `*LinkingContainer*` classes.

Issue is not a show stopper because existing displays continue to work without change as long as one adds all the unresolved displays to the `org.csstudio.opibuilder/opi_search_path`.

kasemir, 2015-02-26:
@ayemos Can you look at this, since you may by now be familiar with the linking container code?

kasemir, 2015-02-26:
Comparing an earlier 4.x with 'master'.

The directory layout is like this:

/CSS/main.opi
/CSS/elsewhere/embedded.opi
/CSS/elsewhere/final.opi

main.opi has a linking container to load embedded.opi.
embedded.opi has an ActionButton to open final.opi. 

For both the older and the current 4.x code, OpenDisplayAction just has "final.opi" as a path.
In OpenDisplayAction.run, ResourceUtil is called to resolve it.

In the older, functional 4.x code, the ActionButtonModel had a DisplayModel parent with path /CSS/elsewhere/embedded.opi, so the final.opi gets resolved to /CSS/elsewhere/final.opi
 
In the latest code, the ActionButtonModel has a LinkingContainerModel parent, and the final.opi gets resolved to /CSS/final.opi, which doesn't exist, and the action fails.

kasemir, 2015-02-26:
Another way to look at this:

Used to be ActionButtonModel -> parent DisplayModel(/CSS/elsewhere/embedded.opi) -> parent DisplayModel(/CSS/main.opi).
The relative path of an action was resolved within the immediate parent, /CSS/elsewhere/embedded.opi.

Now have ActionButtonModel -> parent LinkingContainerModel -> parent DisplayModel(/CSS/main.opi).
The LinkingContainerModel does know about its real DisplayModel(/CSS/elsewhere/embedded.opi), but the ResourceUtil simply follows the parent links, unaware of the LinkingContainerModel because that's in a widget class, not a base opibuilder class.

What might help: Change that into ActionButtonModel -> parent LinkingContainerModel -> parent DisplayModel(/CSS/elsewhere/embedded.opi) -> parent DisplayModel(/CSS/main.opi).
That way, ResourceUtil would again locate the proper parent for resolving paths.

ayemos, 2015-03-02:
@kasemir
Thank you for your report, and maybe the way I sent the last request was not cosy, but anyway I'll try to deal with the issue.
Give me just a few weeks please.

kasemir, 2015-03-02:
@ayemos 
Thanks for the reply, I was wondering if you're still there.
A few weeks is a bit long, since many of our existing screens are broken because of this issue.
Meanwhile, I think I found a patch for AbstractWidgetModel.getRootDisplayModel(). By making it aware of the new AbstractLinkingContainerModel, it can return the correct display model, and all's fine.
Will check a little more and then issue a pull request.

PR #996 (2015-03-02): Bug977
Fixes #977 

#913 performed quite some change on the linking container, introducing a new AbstractLinkingContainerModel and how the linking container appears in the widget tree.
It added several `if (.. instanceof *Linking*)` checks to account for the special handling of the linking container.
This adds one more to handle the resolution of relative links. Stuff within the linked content is again resolved relative to the location of that content.
closed 2015-03-02 by berryma4