ControlSystemStudio/cs-studio #994: Make classic button styles consistent

thomascobb, 2015-03-02:
There seem to be 4 main button widgets in BOY, and they all draw their borders differently. I'd like to rationalise the behaviour, but I'm not sure which is the correct behaviour.

#### Action Button
1 pixel light grey top, 1 pixel grey bottom
[image]

	ButtonScheme BUTTON = new ButtonScheme(
			new Color[] {buttonLightest},
			new Color[] {buttonDarker}
	);

#### Menu Button
Doesn't draw a border. This is what it looks like with the border property set to "Button Raised".

[image]

#### Boolean Button
[image]
3 pixel border in square mode

	private final static int SQURE_BORDER_WIDTH = 3; 
	...
	pattern = GraphicsUtil.createScaledPattern(graphics, Display.getCurrent(), clientArea.x, clientArea.y, clientArea.x, clientArea.y+SQURE_BORDER_WIDTH, GRAY_COLOR, DARK_GRAY_COLOR);

#### Choice Button
1 pixel light grey top, 2 pixels dark grey bottom. Inherits whatever the draw2d ToggleButton does.

[image]

#### Proposed changes
The line of least resistance would appear to be:
* Make the action button draw a border that looks like the "Button Raised" border property
* Leave the menu button as it is
* Reduce the toggle button border to 2 pixels and make the grey colours the same as the "Button Raised" border property
* Leave the choice button as it is

Sound reasonable to everyone?

lcavalli, 2017-02-03:
Hi,
I made a little patch to harmonize the buttons rendering. All of them look like the choice buttons now.
This is the final output (top two boolean buttons, middle a choice button with two options, bottom two active button with toggle option):

[image]

Patch attached for preliminary review. If interested I can open a pull request (next week, I'm leaving the office).

[button_rendering.zip](https://github.com/ControlSystemStudio/cs-studio/files/751044/button_rendering.zip)

kasemir, 2017-02-03:
Go ahead with a pull request.

lcavalli, 2017-02-04:
Created pull request #2068.