ControlSystemStudio/cs-studio #995: DataBrowser: Show names of archived fields

mark0n, 2015-03-02:
It would be great if you could select the field name in addition to the PV name from the DataBrowser search window. Rationale: We are archiving some PVs that provide ~20 fields with names that are hard to remember (yes, motor record). It would be nice if you could pick the field name from the list before sending it to a new DataBrowser plot window. For the best user experience you probably want to pick the PV name first and select the field in a second step (if you don't want to go with the default ".VAL").
This might require a redesign of the UI as well as some thought about how this can be implemented with different back-ends. Murali seems to be willing to make changes to make that happen :-)

kasemir, 2015-03-02:
Don't you already have 99% of that?
If your archive data source has MyMotor, MyMotor.DMOV, .. which are separate Channel Access channels, then they show up in the archive search, in the auto-completion.

mark0n, 2015-03-04:
As far as I understood Murali his Archiver Appliance does not handle fields as separate channels. This might be different from Channel Archiver and the CS-Studio RDB Archiver. If they would be listed as separate channels I could find them by searching for "MyMotor.*" but it seems to be not that easy...

What do you mean by auto-completion? My CS-Studio only seems to auto-complete from the history of manually entered PV names. I'm not aware of auto-complete for unknown PV names/fields (this would require querying the archiver while typing).

kasemir, 2015-03-04:
On Mar 3, 2015, at 7:15 PM, Martin Konrad <notifications@github.com<mailto:notifications@github.com>>
 wrote:

As far as I understood Murali his Archiver Appliance does not handle fields as separate channels. This might be different from Channel Archiver and the CS-Studio RDB Archiver. If they would be listed as separate channels I could find them by searching for "MyMotor.*" but it seems to be not that easy...

What do you mean by auto-completion? My CS-Studio only seems to auto-complete from the history of manually entered PV names. I'm not aware of auto-complete for unknown PV names/fields (this would require querying the archiver while typing).

Autocompletion uses what you tell it to use.
ITER created autocompletion based on archived channels:
https://github.com/ControlSystemStudio/org.csstudio.iter/tree/master/plugins/org.csstudio.autocomplete.archive

I assume you use the channel finder, so you could add your archived channel names to the channel finder as well.

-Kay