ControlSystemStudio/cs-studio #1053: databrowser: Null pointer exception when pv is not in archiver

berryma4, 2015-04-08:
Null pointer exception when pv is not in archiver.

```
2015-04-08 15:41:15.673 WARNING [Thread 67] org.csstudio.trends.databrowser2.ui.Controller$1 (archiveFetchFailed) - No archived data for BSS_CGS:TI_N0033:T_RD
java.lang.NullPointerException
	at org.epics.archiverappliance.retrieval.client.InputStreamBackedGenMsg.fetchData(InputStreamBackedGenMsg.java:96)
	at org.epics.archiverappliance.retrieval.client.InputStreamBackedGenMsg.readAndUnescapeLine(InputStreamBackedGenMsg.java:113)
	at org.epics.archiverappliance.retrieval.client.InputStreamBackedGenMsg.loopInfoLine(InputStreamBackedGenMsg.java:181)
	at org.epics.archiverappliance.retrieval.client.InputStreamBackedGenMsg.readLineAndParseNextMessage(InputStreamBackedGenMsg.java:206)
	at org.epics.archiverappliance.retrieval.client.InputStreamBackedGenMsg.access$000(InputStreamBackedGenMsg.java:35)
	at org.epics.archiverappliance.retrieval.client.InputStreamBackedGenMsg$1.next(InputStreamBackedGenMsg.java:75)
	at org.epics.archiverappliance.retrieval.client.InputStreamBackedGenMsg$1.next(InputStreamBackedGenMsg.java:64)
	at org.csstudio.archive.reader.appliance.ApplianceStatisticsValueIterator.next(ApplianceStatisticsValueIterator.java:135)
	at org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread.run(ArchiveFetchJob.java:129)
	at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

```

berryma4, 2015-04-08:
```
2015-04-08 15:44:47.586 WARNING [Thread 102] org.csstudio.trends.databrowser2.ui.Controller$1 (archiveFetchFailed) - No archived data for BSS_CGS:TI_N0102:T_RD
org.epics.archiverappliance.retrieval.client.RuntimeIOException: Exception near line 803
	at org.epics.archiverappliance.retrieval.client.InputStreamBackedGenMsg$1.next(InputStreamBackedGenMsg.java:78)
	at org.epics.archiverappliance.retrieval.client.InputStreamBackedGenMsg$1.next(InputStreamBackedGenMsg.java:64)
	at org.csstudio.archive.reader.appliance.ApplianceStatisticsValueIterator.next(ApplianceStatisticsValueIterator.java:135)
	at org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread.run(ArchiveFetchJob.java:129)
	at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.io.IOException: Stream closed
	at java.io.BufferedInputStream.getBufIfOpen(Unknown Source)
	at java.io.BufferedInputStream.read(Unknown Source)
	at java.io.FilterInputStream.read(Unknown Source)
	at org.epics.archiverappliance.retrieval.client.InputStreamBackedGenMsg.fetchData(InputStreamBackedGenMsg.java:96)
	at org.epics.archiverappliance.retrieval.client.InputStreamBackedGenMsg.readAndUnescapeLine(InputStreamBackedGenMsg.java:113)
	at org.epics.archiverappliance.retrieval.client.InputStreamBackedGenMsg.loopInfoLine(InputStreamBackedGenMsg.java:181)
	at org.epics.archiverappliance.retrieval.client.InputStreamBackedGenMsg.readLineAndParseNextMessage(InputStreamBackedGenMsg.java:206)
	at org.epics.archiverappliance.retrieval.client.InputStreamBackedGenMsg.access$000(InputStreamBackedGenMsg.java:35)
	at org.epics.archiverappliance.retrieval.client.InputStreamBackedGenMsg$1.next(InputStreamBackedGenMsg.java:75)
	... 8 more
```

kasemir, 2015-04-08:
I assume the issue is in the archiverappliance implementation.
It's very expected that some PVs have no archived data, so this should not be an error.

closed 2015-05-08 by berryma4