ControlSystemStudio/cs-studio #1055: repo format rules

berryma4, 2015-04-10:
Taking this discussion to a issue ticket (from #818 #1043 removing the non existant plugin from application.all.feature ( d794b2261cda368008d14e8d0ee5670a475a77fb )

kunal:
'''
Oh no, I have the git.autocrlf enabled and eclipse configured to ignore whitespace....which is why this file was converted to UNIX endings and was committed without me noticing.

Fixing the all files now would make the history and blame useless...so we should preserve the line endings as they are and agree that all future files should be created with UNIX line endings.

We could add these git attributes to the cs-studio repository itself and thus ensure that this policy is followed...especially by careless people like me.
Additionally we should also define the java code format
currently the repository is a mix of 
1. eclipse convention (tab size=4)
2. java convention (tab size=8)
3. java convention with no tabs (only spaces)
This makes code really difficult to read at times.

@berryma4 @kasemir 
Should this be moved to a different issues?
'''

kasemir, 2015-04-10:
Yes, it might help a little if all contributors had similar Eclipse editor settings:

* Use Unix line endings (http://stackoverflow.com/questions/1886185/eclipse-and-windows-newlines)
* Use spaces instead of tabs (which is in the basic editor settings plus Java/Code Style/Formatter)
* Configure Java/Editor/Save Actions
  1) "Organize imports"
  2) "Additional Actions": Remove trailing white space on all lines

We're unlikely to agree on all those options.

I think we can agree on the basic whitespace handling.
Spaces instead of tabs will result in source code always looking the same, no matter if you prefer 4 or 8 char indentation. Space at end of line is often left behind as you move things around. Order of imports also helps to stay consistent.

Remaining code formatting is a matter of personal preference.
I REALLY prefer balanced braces
````
  int func()
  {
     if (...)
     {
     }
  }
````
over the terribly awful, ugly and unreadable braces left dangling at the end of a line like a forgotten stepchild:
````
  int func()  {
     if (...)  {
     }
  }
````
.. but again that's a personal preference, so even though I'm definitely right on this one, I can't force this on everybody else. Similar, using spaces instead of tabs with indentation by 4 spaces is obviously the best, but if we at least agree on spaces, it'll look the same in each ediror, and your tab-set-at-9 don't look funny with my tabs-set-at-4 (except I don't actually use tabs myself since I prefer spaces).

shroffk, 2015-04-10:
I agree with Kay in regards to discouraging the use of tabs.

I have been using Java Conventions with only spaces, as far as possible I would prefer to be as close to the java conventions as possible. This way people using other IDEs are less likely to modify too much when useing auto-format functions.

willrogers, 2015-04-11:
It would be great to have a consensus on this.  In fact, I think I would vote for a one-off cleanup of the entire repository at the expense of everyone having to use `git diff -w` for evermore.

berryma4, 2015-04-11:
to enforce:
http://docs.sonarqube.org/display/SONAR/SonarQube+in+Eclipse
http://docs.sonarqube.org/display/SONAR/Jenkins+Plugin

maybe we should put format rules together; people can check before commit, and jenkins can also check on pull requests?

also there is checkstyle:
http://checkstyle.sourceforge.net/config_whitespace.html

richardfearn, 2015-04-13:
> Configure Java/Editor/Save Actions 1) "Organize imports" 2) "Additional Actions": Remove trailing white space on all lines

I am against this *unless* trailing whitespace is removed from all files first. Otherwise it just leads to unrelated whitespace changes every time a file is edited.

willrogers, 2015-04-21:
It would be nice to have a consensus on whether we should do a one-off clean-up.  We should definitely set up guidelines.

My view is that if a file is not compliant **now**, in the future, it will either:
  * remain uncompliant or
  * have whitespace changes that we need to compare with `git diff -w`

That is, either we clean up the whole lot or we accept having varying whitespace forever.  I am for the clean-up, but open to arguments against it.  At the moment I keep having to change my Eclipse preferences to comply with whichever file I happen to be editing.

More generally, we should use `git diff -w` if we find whitespace differences in history, but not for comparing the changes we are making, otherwise there's the risk of introducing whitespace errors.

willrogers, 2015-04-23:
Here's a question I asked on Stack Exchange about this topic: http://programmers.stackexchange.com/questions/279849/should-i-edit-a-codebases-whitespace-to-conform-to-a-coding-style.

berryma4, 2015-04-23:
I would vote for the clean up ... but, I believe first we need the enforcement in place.

kasemir, 2015-04-23:
Is all that's left really a chicken-or-egg problem of how to start?

Do we otherwise agree on
* Unix line endings
* Spaces instead of tabs
* 4 space 'tab' width
* remove training white space
?

Then I'd say everybody just do what you can.
Configure your IDE to use Unix line endings: http://stackoverflow.com/questions/1886185/eclipse-and-windows-newlines. Select spaces, 4 of them, instead of tabs, in the basic editor settings plus Java/Code Style/Formatter. Configure Java/Editor/Save Actions, Additional Actions, Remove trailing white space on all lines.

If somebody has spare cycles for a bulk update, say so in here so it can be done once. If nobody steps forward, I'll do that at the code-a-thon.
For the enforcement, at the code-a-thon the plan is to explain/understand/document how the hudson/jenkins setup works. While at it, we can add a checkstyle run for whitespace.

willrogers, 2015-04-24:
I think the code-a-thon is a good time to do this.

I am happy with those conventions.

richardfearn, 2015-04-24:
I'm still against trying to enforce settings (e.g. what line ending to use, or enabling save actions) **before** the files themselves have been cleaned up. But I suppose that since whitespace changes are being accidentally committed now, it's not a big deal for that to continue for a while. Over time things will get better.

However Kay's suggestion does rely on individual developers setting up their IDE correctly. I think that specifying the settings on a per-project basis, and having those settings committed to the repo so everyone uses them automatically, is more likely to give us the results we want.

richardfearn, 2015-04-24:
+1 to the conventions, by the way!

berryma4, 2015-05-18:
code a thon finshed!

closed 2015-05-18 by berryma4

willrogers, 2015-07-08:
@jbobnar did the message get through about the conventions?

There's at least one of your merges that reintroduced some trailing whitespaces: e95684b71ee3168eccd4e98145ba71b6c07d6913

We should try and make sure we don't merge code that violates the conventions, but I think it's still a bit tricky to enforce.