ControlSystemStudio/cs-studio #1058: Many plugins have test classes that aren't in a source folder

richardfearn, 2015-04-13:
e.g. classes in [org.csstudio.autocomplete/test](https://github.com/ControlSystemStudio/cs-studio/tree/master/core/ui/ui-plugins/org.csstudio.autocomplete/test) aren't on the [org.csstudio.autocomplete classpath](https://github.com/ControlSystemStudio/cs-studio/blob/master/core/ui/ui-plugins/org.csstudio.autocomplete/.classpath).

Two problems here:

1. The tests aren't compiled/executed when using `mvn clean verify`.
2. The tests aren't on the build path in Eclipse, so they don't compile in Eclipse, and can't be executed.

Rather than just putting the test directory on the classpath (which would help within Eclipse) it looks like the preferred solution is to move the tests into a `.test` fragment - is that correct?

richardfearn, 2015-04-13:
I think there are 28 plugins like this:

1. `org.csstudio.dal.tine`
1. `org.csstudio.autocomplete`
1. `org.csstudio.ui.util`
1. `org.csstudio.utility.test`
1. `org.csstudio.utility.product`
1. `org.csstudio.security.ui`
1. `org.csstudio.alarm.beast.ui.alarmtree`
1. `org.csstudio.logbook.olog.properties`
1. `org.csstudio.logbook.viewer`
1. `org.csstudio.logbook.ui.extra`
1. `org.csstudio.logbook.ui`
1. `org.csstudio.scan.ui.scantree`
1. `org.csstudio.scan.client`
1. `org.csstudio.scan.ui.plot`
1. `org.csstudio.scan.ui.scanmonitor`
1. `org.csstudio.scan.log`
1. `org.csstudio.shift.viewer`
1. `org.csstudio.opibuilder.adl2boy`
1. `org.csstudio.unitconversion.ui`
1. `org.csstudio.archive.reader.channelarchiver`
1. `org.csstudio.opibuilder.widgets.extra`
1. `org.csstudio.scan.ui.scandata`
1. `org.csstudio.scan.log.derby`
1. `org.csstudio.archive.reader.archiverecord`
1. `org.csstudio.shift.ui`
1. `org.csstudio.logging.jms2rdb`
1. `org.csstudio.utility.adlParser`
1. `org.csstudio.shift`

I don't think any of them have a `.test` fragment.

berryma4, 2015-04-13:
that is correct, all unit tests should be in a test fragment named {plugin}.test
Be careful, some are integration tests that require a server.  As discussed before, these should be updated to use mock objects.

jpenning, 2015-04-13:
>  2. |org.csstudio.autocomplete|
>  3. |org.csstudio.ui.util|
>  4. |org.csstudio.utility.test|
>  5. |org.csstudio.utility.product|
>  6. |org.csstudio.security.ui|
>  7. |org.csstudio.alarm.beast.ui.alarmtree|
>  8. |org.csstudio.logbook.olog.properties|
>  9. |org.csstudio.logbook.viewer|
> 10. |org.csstudio.logbook.ui.extra|
> 11. |org.csstudio.logbook.ui|
> 12. |org.csstudio.scan.ui.scantree|
> 13. |org.csstudio.scan.client|
> 14. |org.csstudio.scan.ui.plot|
> 15. |org.csstudio.scan.ui.scanmonitor|
> 16. |org.csstudio.scan.log|
> 17. |org.csstudio.shift.viewer|
> 18. |org.csstudio.opibuilder.adl2boy|
> 19. |org.csstudio.unitconversion.ui|
> 20. |org.csstudio.archive.reader.channelarchiver|
> 21. |org.csstudio.opibuilder.widgets.extra|
> 22. |org.csstudio.scan.ui.scandata|
> 23. |org.csstudio.scan.log.derby|
> 24. |org.csstudio.archive.reader.archiverecord|
> 25. |org.csstudio.shift.ui|
> 26. |org.csstudio.logging.jms2rdb|
> 27. |org.csstudio.utility.adlParser|
> 28. |org.csstudio.shift|

Hi,

this is Jörg from DESY.

Oh well, the
|  org.csstudio.dal.tine
will probably be used only at DESY (and actually is outdated), therefore 
I don't think it is of much value elsewhere - and may be omitted at all.

We have a lot more plugins which do not currently have a test fragment, 
like they should. But all tests from us at least should have a source 
folder named test.
We will take care of that in some future ... I am afraid not to know 
exactly when.

Jörg.

|

jpenning, 2015-04-13:
Hi, this is Jörg from DESY.

Well, the
|  org.csstudio.dal.tine
|will probably be used only at DESY, therefore I don't think it is of 
much value elsewhere - and may be omitted at all.

We have a lot more plugins which do not have a test fragment, like they 
should. But all tests at least should live in a source folder named test 
(sometimes test/java).

Jörg.

richardfearn, 2015-04-21:
`org.csstudio.opibuilder.adl2boy` is tricky to deal with - test code in there depends on test code (`ADLTestObjects.java`) in `org.csstudio.utility.adlParser`, and a test fragment for `adl2boy` cannot depend on a test fragment for `adlParser`. Will come back to those two.

richardfearn, 2015-04-29:
http://www.modumind.com/2007/06/20/unit-testing-plug-ins-with-fragments/

willrogers, 2015-07-08:
Looking at opibuilder, I find that `*Test.java` finds files 

* `org.csstudio.swt.widgets.test`
* `org.csstudio.opibuilder.test`
* `org.csstudio.opibuilder.converter`
* `org.csstudio.opibuilder.adl2boy`  

The first two are correct except that their `org.csstudio.opibuilder.test/pom.xml` has `<packaging>eclipse-plugin</packaging>` so maven isn't running the tests.

The third is mainly DLS's concern at present so I'll have a look at this (Richard, did you miss this one?).

The last was mentioned above.

richardfearn, 2015-07-23:
> The third is mainly DLS's concern at present so I'll have a look at this (Richard, did you miss this one?).

Sorry, I have been on paternity leave for a while :-)

Yes, I did miss that one. My script was looking for a `test` folder in the top level of each plugin. In {{o.c.opibuilder.converter}} the test code was in `src/test`.

Thanks for moving the test code in e6c1c27!

richardfearn, 2015-07-23:
Updated list, as of 2015-07-23:

1. `org.csstudio.archive.reader.archiverecord`
1. `org.csstudio.archive.reader.channelarchiver`
1. `org.csstudio.logbook.ui`
1. `org.csstudio.logging.jms2rdb`
1. `org.csstudio.opibuilder.adl2boy`
1. `org.csstudio.opibuilder.widgets.extra`
1. `org.csstudio.scan.client`
1. `org.csstudio.scan.log`
1. `org.csstudio.scan.log.derby`
1. `org.csstudio.scan.ui.plot`
1. `org.csstudio.scan.ui.scandata`
1. `org.csstudio.scan.ui.scanmonitor`
1. `org.csstudio.scan.ui.scantree`
1. `org.csstudio.shift`
1. `org.csstudio.shift.ui`
1. `org.csstudio.shift.viewer`
1. `org.csstudio.unitconversion.ui`
1. `org.csstudio.utility.adlParser`