ControlSystemStudio/cs-studio #1060: Many plugins use SLF4J for logging

richardfearn, 2015-04-13:
I thought CS-Studio had standardised on JUL for logging. But quite a few plugins use SLF4J:

1. `org.csstudio.dct`
1. `org.csstudio.dct.ui`
1. `org.csstudio.desy.logging.ui`
1. `org.csstudio.domain.common`
1. `org.csstudio.platform.ldapAuthorization`
1. `org.csstudio.platform.simpledal`
1. `org.csstudio.remote`
1. `org.csstudio.sds`
1. `org.csstudio.sds.components.ui`
1. `org.csstudio.sds.importer.ui`
1. `org.csstudio.sds.samples`
1. `org.csstudio.sds.service.pvvalidation.mkk`
1. `org.csstudio.sds.settings.desy`
1. `org.csstudio.sds.ui`
1. `org.csstudio.servicelocator`
1. `org.csstudio.testsuite`
1. `org.csstudio.utility.adlConverter`
1. `org.csstudio.utility.caSnooperUi`
1. `org.csstudio.utility.ldap.nameSpaceBrowser`
1. `org.csstudio.utility.ldap.service`
1. `org.csstudio.utility.ldap.service.impl`
1. `org.csstudio.utility.ldap.treeconfiguration`
1. `org.csstudio.utility.nameSpaceSearch`
1. `org.csstudio.utility.quickstart`
1. `org.csstudio.utility.screenshot`
1. `org.csstudio.utility.tine`
1. `org.csstudio.utility.treemodel`

Should they be converted to use JUL?

jpenning, 2015-04-13:
>  2. |org.csstudio.dct.ui|
>  3. |org.csstudio.desy.logging.ui|
>  4. |org.csstudio.domain.common|
>  5. |org.csstudio.platform.ldapAuthorization|
>  6. |org.csstudio.platform.simpledal|
>  7. |org.csstudio.remote|
>  8. |org.csstudio.sds|
>  9. |org.csstudio.sds.components.ui|
> 10. |org.csstudio.sds.importer.ui|
> 11. |org.csstudio.sds.samples|
> 12. |org.csstudio.sds.service.pvvalidation.mkk|
> 13. |org.csstudio.sds.settings.desy|
> 14. |org.csstudio.sds.ui|
> 15. |org.csstudio.servicelocator|
> 16. |org.csstudio.testsuite|
> 17. |org.csstudio.utility.adlConverter|
> 18. |org.csstudio.utility.caSnooperUi|
> 19. |org.csstudio.utility.ldap.nameSpaceBrowser|
> 20. |org.csstudio.utility.ldap.service|
> 21. |org.csstudio.utility.ldap.service.impl|
> 22. |org.csstudio.utility.ldap.treeconfiguration|
> 23. |org.csstudio.utility.nameSpaceSearch|
> 24. |org.csstudio.utility.quickstart|
> 25. |org.csstudio.utility.screenshot|
> 26. |org.csstudio.utility.tine|
> 27. |org.csstudio.utility.treemodel|

This is all DESY stuff, we did not convert to JUL when it was introduced 
in CSS.
Currently we are not building our products from the community edition, 
so it does not matter directly to us.
I guess only few of the given plugins are used outside DESY ( DCT and 
SNL being candidates) - so converting might be a wasted effort.

Perhaps it would be best to ask who actually is using things and keep 
only the relevant parts in the community edition.

I'm sorry to be of no more help, but currently we are mainly dealing 
with our CSS-based tools for developing the XFEL accelerator and leave 
everything else as it is.

Jörg.

willrogers, 2016-04-06:
There should be no further references to slf4j.

closed 2016-04-06 by willrogers