ControlSystemStudio/cs-studio #1062: /BOY Examples/Miscellaneous/DynamicLoadWidgets is broken

kasemir, 2015-04-15:
The loaded widgets are strangely resized.

kasemir, 2015-04-15:
Seems related to commits around f87422e53f5b1f011c630dedf52a3d44e6097726 when the LinkingContainer options "zoom_to_fit" and "auto_size" where merged into "resize_behaviour".

In the example script, replacing
```
linkingContainer.setPropertyValue("auto_size", true);
linkingContainer.setPropertyValue("zoom_to_fit", false);
```

with 
```
linkingContainer.setPropertyValue("resize_behaviour", 1);
```
seems to work.

kasemir, 2015-04-15:
@mfurseman : The  LinkingContainer widget still understands the original options "zoom_to_fit" and "auto_size", but behaves differently when these are set from scripts.

I changed the example in 7b50ee10375d5d068be5aadedf400b592f1ed1ff.
Existing scripts in production will need to be updated accordingly.

Not sure if there's a simple way in the LinkingContainerModel to catch calls to `setPropertyValue(..)` for the older properties, which would then instead update the new "resize_behaviour".

mfurseman, 2015-04-16:
As this is closed should I assume that it no longer needs attention?

I haven't had a chance to look at this yet, but if it looks like many scripts rely on this functionality I expect we can find a way to get the deprecated calls to `setPropertyValue(..) ` updating correctly as Kay suggested.

berryma4, 2015-04-16:
The merge closed the issue.  It should stay open for the backward compat.

kasemir, 2015-04-16:
Matthew, if you have time to do this, I think it'll be best to update the LinkingContainerModel to stay compatible. After thinking about this for 5 minutes, a solution might be to overide setPropertyValue() in LinkingContainerModel so that it still handles the old properties:
```
	@Override
       public void setPropertyValue(Object id, Object value)
	{
	    // Route updates to deprecated properties to the new one 
	    if (PROP_ZOOMTOFITALL.equals(id))
	    {
	        super.setPropertyValue(PROP_RESIZE_BEHAVIOUR, /* somehow transform 'value' */);
	    }
	    else if (PROP_AUTO_SIZE.equals(id))
        {
            super.setPropertyValue(PROP_RESIZE_BEHAVIOUR, /* somehow transform 'value' */);
        }
	    else
	        super.setPropertyValue(id, value);
    }
```

Maybe with added log message "Dude, you need to update your script, because .."

kasemir, 2015-04-16:
Something else that would have helped with the displays that we have deployed: Just change the default.

Right now LinkingContainerModel "resize_behaviour" defaults to SIZE_OPI_TO_CONTAINER. Scripts changing the older  ZOOMTOFITALL & AUTO_SIZE had no effect, and because of the way things resized, you ended up with tiny unreadable stuff.

A default of CROP_OPI or SCROLL_OPI would have kept the linked content at its original size. It wouldn't look pretty, but you could have used it.

mfurseman, 2015-04-17:
The current default was chosen as it is the same as the previous default:
`PROP_ZOOMTOFITALL -> true`
`PROP_AUTO_SIZE ->  false`

I'll see if I can get a pull request for this soon.

mfurseman, 2015-04-24:
I've made two (identical) commits to fix this on master and 4.0.x. The only test case I've seen for this is the BOY example, so if this doesn't work for some other scripts I can look into fixing it more robustly.

I except I would find an answer to this in the documentation if I looked hard enough: If DLS want bug fixes on 4.0.x and on master, would it be better to make them only on 4.0.x, and then open a pull request to merge 4.0.x into master?

berryma4, 2015-04-24:
I try my best to make sure any bug fixes on 4.0.x make it to master.  Though, the work flow of pull request to 4.0.x, then pull request from 4.0.x to master (instead of separate commits) is easier to maintain and make sure everything moves forward.

kasemir, 2015-04-24:
Thanks, I think this fixes the problem!

PR #1063 (2015-04-16): o.c.opibuilder: Update Miscellaneous/DynamicLoadWidgets
.. because LinkingContainer has changed.
Fixes #1062, i.e. the example works again.
Production scripts implemented following that example will also need to be updated.
closed 2015-04-24 by kasemir