ControlSystemStudio/cs-studio #1064: Update to Eclipse 4.5 Mars

kasemir, 2015-04-16:
Eclipse 4.5 aka Mars is expected end of June 2015.

CS-Studio 'master' should then be updated to successfully build against it.

Things to gain for CSS:
 *  Customize Perspective Dialog is back (ITER had missed it).
 * UI responsiveness monitoring. We should include & enable this, which will put information about UI freezes, for example code that accesses files on the UI thread, into the log, so we can then fix frequent problems.
 * org.eclipse.core.runtime.jobs.JobGroup helps with creating jobs that use all CPUs (can also be done with Executors.newWorkStealingPool() if connection to Progress View is not required)
 * SWT on Linux can use GTK+ 3 (I see org.eclipse.swt.internal.gtk.version=2.24.23 in Help, About, Installation Detail, Configuration on our Linux computers, so that GTK 2 is also still supported).
 *  JFace StructuredViewer has IStructuredSelection getStructuredSelection(), no more casts.
 * In addition to the plugin_customization.ini, products can have a *.pde file with saved preferences.

Things to gain in the IDE:
 * Nested/Hierarchical view of projects that might better reflect how we now structure things in the repository
* Target platform "Reload" may help those who use remote repositories

For more: https://www.eclipse.org/eclipse/news/4.5/M5, which ends in links to ../M4, M2, ..

berryma4, 2015-10-05:
I have also removed 32-bit builds

berryma4, 2015-10-07:
0.23.1 tycho?

closed 2015-10-14 by berryma4