ControlSystemStudio/cs-studio #1065: OPI Shell/View: Enable shell support based on preferences?

kasemir, 2015-04-16:
Related to #544:

Should the support for OPI Display in "Standalone Window", i.e. `OPIShell`, be enabled/disabled based on preferences?

If yes, as has been mentioned in the parent ticket, how to do that?

It appears in the GUI as
 1) Navigator support "Open With", "OPI Display (Standalone Window)"
 2) Context menu of actions
 3) In OPI Editor as one of the properties of the `OpenDisplayAction`

In 3, there may not be a need to remove it. "Standalone window" remains available at edit time, after all you could receive displays from other sites that do use the "Standalone window" option. At runtime, any requests for a new Shell will instead open a new Window.

For 2), the context menu is created in `OPIRunnerContextMenuProvider.addWidgetActionToMenu`, where it can be suppressed if preferences are set to disable `OPIShell` support.

1) is difficult, because it is based on this text in the o.c.opibuilder/plugin.xml:
```
   <extension point="org.eclipse.ui.editors">
      <editor
            default="false"
            extensions="opi"
            icon="icons/OPIRunner.png"
            id="org.csstudio.opibuilder.ShellLauncher"
            launcher="org.csstudio.opibuilder.runmode.ShellLauncher"
            name="OPI Display (Standalone Window)">
      </editor>
   </extension>
```

One hack I tried:
Move from the plugin.xml into a separate plugin_add.xml, than do this during opibuilder plugin startup:
```
final IExtensionRegistry registry = Platform.getExtensionRegistry();
if (PreferencesHelper.isOpiShellEnabled())
{   // Find 'contributor' associated with this plugin
    IContributor contrib = null;
    for (IExtension extension : registry.getExtensionPoint("org.eclipse.ui.editors").getExtensions())
    {
        if ("org.csstudio.opibuilder".equals(extension.getNamespaceIdentifier()))
        {
            contrib = extension.getContributor();
            break;
        }
    }
    // Load optional additions to plugin.xml
    final Bundle bundle = plugin.getBundle();
    final InputStream stream = FileLocator.openStream(bundle, new Path("plugin_add.xml"), false);
    registry.addContribution(stream, contrib, false, "OPIShell support", null, null);
}

// Or later remove the addition again as the preference changes:
if (PreferencesHelper.isOpiShellEnabled() == false)
    for (IConfigurationElement config : registry.getConfigurationElementsFor("org.eclipse.ui.editors"))
    {
        if ("org.csstudio.opibuilder.ShellLauncher".equals(config.getAttribute("id")))
        {
            registry.removeExtension(config.getDeclaringExtension(), null);
            break;
        }
    }
```

Such changes to the registry are only possible when the product is run with `-Declipse.registry.nulltoken=true`.

Another way might be  http://wiki.eclipse.org/Product_Customization. It's a way to add XML transformations to a product which would for example remove the 
```
<extension point="org.eclipse.ui.editors">
      <editor id="org.csstudio.opibuilder.ShellLauncher"
```
from the plugin.xml at startup. It does not, however, allow for changes of the registry at runtime.

Another way which I haven't looked into might be somehow hooking into the "Open With>" context menu, where is could be possible to add/remove an object contribution that enables/disables based on a preference.

willrogers, 2015-04-21:
We have agreed in #544 to postpone this ticket until there is a clear need.

However, I note that we have a similar problem elsewhere: if editing OPIs is disabled in the preferences, the option remains in the navigator to open the OPI editor.

kasemir, 2015-04-23:
You're right. The way that was handled (internally close the editor right away and instead start a runtime) was also a hack, but one that's visible to the end user.

That option, however, is misguided anyway. Why select no-edit?
To keep users from accidentally damaging displays? -> Use file permissions.
To keep users from creating their own displays? -> Maybe consider putting their energy to use, instead of annoying them.

If the 'no-edit' option was worth doing, it'd be worth doing right, by programmatically adding/removing the Open-in-Editor support for the navigator. But is it?

willrogers, 2015-04-24:
I'll pass this question on to @thomascobb and @gilesknap, who have beamline users that tend to experiment with things.

gilesknap, 2015-04-24:
I would vote to keep the no-edit option. The majority of our users are not power users and we would like to be able to restrict the behaviour of CSS to only present them with the screens they need to do their job and not confuse them with additional views that are aimed at developers.

At Diamond, EDM always runs in read-only mode on beamlines. If users want changes to screens then they ask their controls team representative to implement it. This can then be done to our standards and in a way that will not break with future upgrades to support modules.

For power users I would argue that it would be frustrating to be able to edit screens and then not save them. They would have to use save-as and then not be able to link to the new OPI from any of their existing read-only OPIs.

This is not a show stopper for us, we could live without no-edit mode. However I think it is common practice for such tools to have a developer mode and runtime mode.

closed 2015-05-18 by berryma4