ControlSystemStudio/cs-studio #1071: CAJ 1.1.15 connection issues,

shroffk, 2015-04-20:
With the upgrade to caj 1.1.15 from caj 1.1.14 there is an issue with establishing connections.
It seems like there might be attempt to use a connection is either before it is completely created or connections are prematurely closed.

Will follow up with @msekoranja to determine if there is a change in the way pv connections are established in caj 1.1.15

It seems that there is a race condition introduced which makes reproducing the exact problem difficult. Below is the stack trace of the most frequent manifestation of this issue.

```

2015-04-20 12:15:23.528 WARNING [Thread 56] org.csstudio.simplepv.pvmanager.PVManagerPV (stop) - PV =(indexOf('SR:C01-MG{PS:QL1A}Sum1-Sts')==1?2:indexOf('SR:C01-MG{PS:QL1A}CtctON-Sts')==0?1:0) has already been stopped or was not started yet.
2015-04-20 12:15:23.529 WARNING [Thread 56] org.csstudio.simplepv.pvmanager.PVManagerPV (stop) - PV =(indexOf('SR:C01-MG{PS:QH3B}Sum1-Sts')==1?2:indexOf('SR:C01-MG{PS:QH3B}CtctON-Sts')==0?1:0) has already been stopped or was not started yet.

2015-04-20 12:15:24.545 WARNING [Thread 80] org.csstudio.opibuilder.util.ErrorHandlerUtil (handleError) - Error from pv connection layer: 
java.lang.IllegalStateException: ConnectionCollector for 'SR:C12-MG{PS:QL1B}CtctON-Sts' was closed.
	at org.epics.pvmanager.ConnectionCollector$ConnectionWriteFunction.writeValue(ConnectionCollector.java:45)
	at org.epics.pvmanager.ConnectionCollector$ConnectionWriteFunction.writeValue(ConnectionCollector.java:31)
	at org.epics.pvmanager.MultiplexedChannelHandler$MonitorHandler.processConnection(MultiplexedChannelHandler.java:67)
	at org.epics.pvmanager.MultiplexedChannelHandler.reportConnectionStatus(MultiplexedChannelHandler.java:123)
	at org.epics.pvmanager.MultiplexedChannelHandler.setConnected(MultiplexedChannelHandler.java:364)
	at org.epics.pvmanager.MultiplexedChannelHandler.processConnection(MultiplexedChannelHandler.java:163)
	at org.epics.pvmanager.jca.JCAChannelHandler.access$600(JCAChannelHandler.java:51)
	at org.epics.pvmanager.jca.JCAChannelHandler$3.connectionChanged(JCAChannelHandler.java:423)
	at gov.aps.jca.event.DirectEventDispatcher.dispatch(DirectEventDispatcher.java:56)
	at com.cosylab.epics.caj.CAJChannel.setConnectionState(CAJChannel.java:643)
	at com.cosylab.epics.caj.CAJChannel.connectionCompleted(CAJChannel.java:353)
	at com.cosylab.epics.caj.impl.handlers.CreateChannelResponse.internalHandleResponse(CreateChannelResponse.java:48)
	at com.cosylab.epics.caj.impl.handlers.AbstractCAResponseHandler.handleResponse(AbstractCAResponseHandler.java:110)
	at com.cosylab.epics.caj.impl.CAResponseHandler.handleResponse(CAResponseHandler.java:139)
	at com.cosylab.epics.caj.impl.CATransport.processRead(CATransport.java:511)
	at com.cosylab.epics.caj.impl.CATransport.processRead(CATransport.java:423)
	at com.cosylab.epics.caj.impl.CATransport.handleEvent(CATransport.java:361)
	at com.cosylab.epics.caj.impl.reactor.lf.LeaderFollowersHandler.handleEvent(LeaderFollowersHandler.java:77)
	at com.cosylab.epics.caj.impl.reactor.Reactor.processInternal(Reactor.java:400)
	at com.cosylab.epics.caj.impl.reactor.Reactor.process(Reactor.java:284)
	at com.cosylab.epics.caj.impl.reactor.lf.LeaderFollowersHandler.run(LeaderFollowersHandler.java:91)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

```

msekoranja, 2015-04-20:
The stack trace says: CAJ received create channel response from the server and fired connectionChanged event, the following code ConnectionWriteFunction.writeValue() throws an exception.
(Note: ConnectionWriteFunction.counter field is not synchronized, I hope it's called with some kind of lock held.)

shroffk, 2015-04-20:
The three methods that modify the counter are in fact called with a lock.

kasemir, 2015-04-23:
Can you tell if the actual problem is in CAJ, PVManager, or the way the opibuilder uses the PVManager?
If you have a display without formulas you can try running with vtype.pv as the connection layer, see if the CAJ update results in a problem for that data path as well.

shroffk, 2015-05-06:
#1103

closed 2015-05-06 by shroffk

HongYi-Shern, 2019-10-27:
Hello, I am an epics rookie. Can you give me a case of caj use? Thank you!