ControlSystemStudio/cs-studio #1073: BOY: Add icon to Menu Button

mfurseman, 2015-04-22:
When controlling an Enum PV in EDM the menu button has a rectangular icon hinting to users that it will open a drop down list:
[image]
We would like this functionality in CS-Studio, however the menu button does not provide the option to choose an icon. After looking at implementing this it seems that there would be few differences remaining between the MenuButtonFigure and ActionButtonFigure. 

Although it is relatively easy to just add an icon to the menu button figure, does it seem like unifying these figure classes makes sense?

kasemir, 2015-04-22:
If you can merge the two button figures, I'd say go for it.

As for the specific use with enums and drop-downs, I agree that a visual hint is good.
Do you want to automate this, i.e. menu buttons would always have that icon?

I wouldn't want to limit us to the Motif 'square'. Even past EDM users should be OK with a more common 'drop down' hint, for example the one used by SWT:
[image]  Or goggle for 'drop down menu' images.

mfurseman, 2015-04-23:
I'm not a fan of the Motif square ethier!

We could either choose a graphic to bundle and display on the button (perhaps with an option in the model to hide/show it), or just let the user choose an icon to include as the Action Button does. In the interest of presenting the user with a consistent interface I would prefer to always show a fixed icon, but having created this issue ticket I am sure to be biased.

berryma4, 2015-04-23:
:+1: for the google image (a copyright safe version at least)

kasemir, 2015-04-23:
I'd vote for 'fixed' icon, too. As for the specific icon, I'd either use

a) What SWT uses on the platform for the SWT.ARROW button, Combo drop-down etc. See
https://www.eclipse.org/swt/widgets/ That makes it consistent with the overall look of everything else on that platform. .. but it will result in a different icon for Windows, Linux, Mac OS X. Plus for people who really want everything to look like EDM: If you resurrect the build of SWT for Motif, it'll look 100% like EDM.

b) Some generic 'down pointing triangle' icon like 
[image] Won't match any of the existing toolkits, but that's exactly the point, to have one icon thats used on all platforms.

What I wouldn't like is to use for example the "Windows" drop-down icon and then have that look out of place on Linux and vice versa.

willrogers, 2015-04-24:
There's a little wrinkle here in that the merged figure will slightly change the appearance of the menu button (which was previously a {{Label}} drawn to look like a button).  Is this a problem?

kasemir, 2015-04-24:
There are two fundamentally different standpoints.

The display is pixel-accurate: The shading, cornering needs to stay.

The display describes meaning: You put down a TextUpdate with PV name and alarm sensitive border; a MenuButton with 4 entries. Exact look changes between windows, linux, os x right now, and will again change with Java FX or a better web display in the future. On one platform all buttons may have rounded corners, on the other sharp edges. That's actually desired, so it matches the platform.

What I want for the display is to follow the 'meaning' idea. Have something that is meant to act like a button look like a button shouldn't then be a problem at all.

closed 2015-11-26 by berryma4