ControlSystemStudio/cs-studio #1081: Macros substitution for linking containers broken

shroffk, 2015-04-24:
On the master branch changes to the linking container have disabled the correct loading of the macros from the file being linked to.
Additionally system macro are also not correctly attached to the linking container( ${LCID} does not work)

In the LinkingContainerEditpart in the configureDisaplyModel(), we see that the macroMap for the display model is correctly updated and has all the required macros, with expected values.

opi files to recreate this issue have been added to the opi-regression-tests repository.

kasemir, 2015-04-24:
Does this also go back to #913?
After the path and script lookups #977 and #998 this would be yet another bug from the Linking container update.

shroffk, 2015-04-24:
It would seem like this could quite probably be the change...since it has significantly changed the way the container is loaded.

commit cf2b6a6c (2015-05-05): bug fix #1081 macro substitution
commit fd130045 (2015-05-05): Merge pull request #1092 from berryma4/master

bug fix #1081 macro substitution
closed 2015-05-05 by berryma4