ControlSystemStudio/cs-studio #1086: Update linked resources when OPI is opened

willrogers, 2015-04-27:
We have a very specific problem.

We are using the `-share_link` option to set up projects in CS-Studio to match the layout on our production filesystem.  This works very well - for example, you can change one of the links to point to a development version, and other links to related displays or scripts still resolve.  Furthermore, we're using our existing launcher program to start OPI files when CS-Studio is already running.

There's one hitch.  In order for this to work correctly, the links in the projects need to be updated when you run an OPI file after CS-Studio is started, but the `-share_link` option is recognised only on CS-Studio startup.

I've fixed this to work at Diamond, but the solution is not elegant.  The more general description of the problem is that we need extra information to pass into OPI builder via `--launcher.openFile`.  The CS-Studio release interprets both filename and macro.  I now interpret filename, macro and shared links.  I also 'borrow' a macro to determine whether or not to start an OPI shell.  In future, we may want to select a particular perspective or start a new workbench window.

In practice, I think we need command-line arguments accepted by OPI builder itself rather than just the ones that Eclipse offers, but I'd like some agreement before starting to figure out the best way to do this.  This would all be invisible to 'normal' users, and would be contained in wrapper scripts for us.

Any thoughts?

kasemir, 2015-04-27:
Why do you need to modify the linked resources at runtime?
Are you switching certain links from a production setup to development directories?
What about *.opi files that have already been opened using the previous resources?

If you really think you need this locally, maybe use symlinks on the file system and only point the CSS workspace to a top-level directory, so it's otherwise unaware of those soft links?

willrogers, 2015-04-27:
1. It's the most seamless way to do it.  When a new version is released, the link will update to the newer version and most users won't even notice.  Also, users get projects containing the screens they use added to their workspace over time.  Otherwise, we'd have to add all projects to everyone's workspace the first time they start CSS.
2. Yes.  This is easier to set up too.  You launch the screen (which sets up the links and the project).  Then when you want to edit a screen, you can change one or more of the links to a dev version and then rename the project.  Relative links still work  Next time you open the screen externally, a new 'prod' project is created, and your 'dev' project is still there too.
3. Good question.
4. I ***think*** that this would be more difficult to maintain, or at least I'd have to think of a way of doing it.  Maybe have a set of projects somewhere on a filesystem that are made up of symlinks.  Users go and import those projects when they start using CSS.  A new release would update those links.  But then you have to make your own version of those links, otherwise changing it for you will change it for everyone else.

willrogers, 2015-04-30:
OK, it sounds like this feature is not very 'official'.  Let me show how we're using it.
[image]

Whenever you open a screen from our 'launcher', as well as popping up a new window it adds the appropriate project to CSS.  'diagOpi_2-46' is the released version, and all the links direct to different locations on the read-only filesystem.  Other projects may link to the same or different versions of those projects.

To do development, edit any of those links to point to a dev version.  In this example the link to 'diagOpi' is broken in the (dev) project for that reason.

It all seems to work very nicely.

willrogers, 2015-04-30:
The filesystem soft link version would be similar, but a bit more manual.  When you open a new screen, first you would have to explicitly create a project containing soft links (and persuade Eclipse that it exists). 

To develop, you'd have to change the links on the filesystem, as CSS doesn't know about them.  Also, if you copy the project from within CSS, it resolves the links and copies all the files.

-----
Alternatively, you have a way of setting everyone's workspace up when they first use CSS.  This has a potential disadvantage of a lot of projects for everyone.  Changing workspaces may also be more difficult.

willrogers, 2015-07-06:
See #1204.

willrogers, 2016-09-23:
This is done, please close.

willrogers, 2016-09-23:
No, I can close this one.

closed 2016-09-23 by willrogers