ControlSystemStudio/cs-studio #1088: Legend for databrowser plot

shroffk, 2015-04-29:
With the new databrowser plot, where the legend is no longer included and the information is displayed on the Axis, certain configurations for the plot do not render well.

An example is show below:
[image]

In this case in addition to maximizing the plot area, since these plots are primarily used to display time series data, it is to provide a way to maximize the horizontal span too.

Proposed solution:
To add a collapsible legend.
[image]

kasemir, 2015-04-29:
Basically, makes sense.

Missing/todo:
 * Add [x] Show legend to properties, persist & reload to/from config file.
 * Instead of its own gray, use plot's background color
 * Add LegendFont (so far: Title, Label, Scale)
 * Context menu "Hide Legend", "Show Legend" (renamed from "..bar") still necessary in spite of properties. Show/hide of legend will 'dirty' the editor (unless configured to ignore 'dirty' changes)
 * Code is/remains mostly in RTPlot

Later:
 * Instead of Composite/Labels/GridLayout, make this an RTPlot PlotPart that's drawn in background thread. No immediate visible difference to users, but better in the long run: Draws in background, can create/release colors instead of having to pass colors to Label, and will allow to somehow reflect which axis contains which trace.

shroffk, 2015-04-30:
Missing/todo:
* legend show update when trace display char are changed (name, color...)

kasemir, 2015-05-05:
Re: legend show update when trace display char are changed (name, color...)
Is now the case.

Re: databrowser.opiwidget should have show/hide toolbar and legend options.
Yes, at runtime, as you had already added.

Re:  .opi visibility properties should override the .plt properties for embedded plots.
Maybe not.
The toolbar visibility property was in the opiwidget because the *.plt failed to preserve that.
Now the *.plt stores the toolbar visibility (thanks to you), but unfortunately we already have opiwidgets storing it as well, so yes, the *opi needs to override the toolbar visibility from the *.plt.
For the legend, I think we start out right: The *.plt file has the config, so the *.opi doesn't need it.

shroffk, 2015-05-06:
Allowing opi widgets to configure toolbar and legend visibility would help with reusing .plt files. The same file can be used in a overview screen where the scarcity of space might make hiding the legend the preferable default option. On a more detailed screen it might make more sense to have the legend visible by default. If the configuration is confined to the .plt file we might need 2 copies of the same - one with visibility set to true one in which it is disabled.

kasemir, 2015-05-06:
OK, let me put the opiwidget option to show/hide legend, overriding that setting from the *.plt model, back in.

closed 2015-05-18 by berryma4