ControlSystemStudio/cs-studio #1094: Consolidate alarm icons

kasemir, 2015-05-01:
Recent update added alarm table icons. The alarm tree always had icons, and now that's inconsistent with the alarm table icons. Alarm tree has the added requirement to reflect if some alarm trigger PVs are disabled. Need to decide how to reflect that with the new icons.

utzeln, 2015-05-04:
The disabled icon state could be a dimmed version of the enabled icons - invalid, major and minor, supposing that even if the alarm is disabled, the alarm server maintains its state.
Ref. https://wiki.eclipse.org/UI_Graphics_:_Design_:_States

kasemir, 2015-05-04:
Note that the disablement can be partial.
Individual alarm trigger PVs are either fully enabled or disabled.
At other levels in the alarm tree (area, system), the icon currently indicates if one or more trigger PVs below that level are disabled. A gray background to indicate "something is disabled" while otherwise using the original icon coloring to reflect the summed state of the remaining PVs might work.

jbobnar, 2015-05-11:
Here are some ideas for the icons. Please let me know, what is acceptable.

Disabled items: 
Currently all icons have white background. We could mark the nodes that have disabled children with gray background.

Half-half icons: 

a) 
We invent icons for each and every combination of severity and latched severity. In that case we can use different combinations of the red cross, yellow triangle, checkbox. For example:
[image]
Downsides:
Users would need to get familiar with lots of new icons. Not a big deal for newcomers, but could be an issue with established users. The 16px icon is also too 
small for displaying that level of detail.

b)
We show double icons, where one icon is for the latched severity and one for the current one. If they are the same or if they fall into one of the categories, for which we already have an icon (e.g. cleared but not acknowledged), we show just one icon.
[image]
[image]
Downsides:
Maybe too colorful and more space is needed to show the tree (in case of a deep tree).

c)
We continue to use the circles and half circles in the tree. The good thing is that users are already used to this and the colors resemble the severity.

kasemir, 2015-05-11:
A gray background to indicate disabled trigger PVs sounds straight forward.

Using the existing color indicators, option c, _everywhere_, doesn't sound too bad to me, except for the color blind. .. Who could then fall back to the text description of current and latched alarm state, which is there after all.

Otherwise, option a, that is a new set of icons since the existing set of recently added table icons fails to represent all possible alarm state combination, is certainly the best option. In theory. In practice, I'm not sure what those icons should look like. Need to reflect the combined alarm state, be easy to identify, as the colored circles were, but in addition support the color blind, so need to use shapes instead of just colors, and then you up with shape overload?

utzeln, 2015-05-12:
Option c everywhere is also fine for me.