ControlSystemStudio/cs-studio #1102: Error message for bad const() PV syntax could be improved.

MichaelRitzert, 2015-05-06:
Currently, the message is

> 2015-05-06 14:08:49 ERROR: Error from pv connection layer: 
> java.lang.IllegalArgumentException: Wrong syntax. Correct examples: const(3.14), const("Bob"), const(1,2,3), const("ON", "OFF")

This could show which PV is the bad one. It took me a while to find it in this case.

MichaelRitzert, 2015-05-06:
I should say, I have a patch for CSS before the big reorganization, but I cannot find where the plugin went, now that I want to create a pull request.

The idea is to allow %s in the error message passed to FunctionParser.parseFunctionWithScalarOrArrayArguments, to use
> throw new IllegalArgumentException(String.format(errorMessage, string));

and change the ConstantChannelHandler constructor to call

> List<Object> tokens = FunctionParser.parseFunctionWithScalarOrArrayArguments(channelName, "Wrong syntax: '%s'. Correct examples: const(3.14), const(\"Bob\"), const(1,2,3), const(\"ON\", \"OFF\"");

kasemir, 2015-05-06:
FunctionParser is in org.csstudio.utility.pvmanager. It's imported from the PVManager. When working on CSS, you only see the binary class file in the IDE, I can't debug into it let alone edit.
Changes to the PVManager would have to be submitted to the PVManager (diirt) sources, and eventually CSS will then get the update.