ControlSystemStudio/cs-studio #1103: Connection from Linking container

shroffk, 2015-05-06:
The modification to the linking container made in the master branch there seem to be significant error with handling connections.

kasemir, 2015-05-06:
Likely caused by #912: Linking container is broken, requiring additional `if (widget instance of LinkingContainer)` type checks when starting/stopping PVs for all widgets.

jbobnar, 2015-05-28:
2a and 2b are bugs in PVManager and PVManagerPV. They are not limited to the LinkingContainer. They are visible also if you use the child opi directly, just less frequently. Hit the refresh button like crazy and you're bound to see it happen.

2b was partially fixed by #1167. The org.csstudio.simplepv.pvmanager has incorrect "synchronization" of start/stop methods. From the git comments I see some deadlocks were being fixed this way, so I do not dare to change this. But here are my thoughts:
- PVManager#start and PVManager#stop should be synchronized, because they are not thread safe. There is a (small) window when both methods could be executed synchronously. Luckily they are always called from the same thread, so a problem like that cannot happen.
- Shouldn't the stop method return after it puts a "stopping" runnable on the notification thread?
- If the notification thread is not a single thread executor, strange things might happen.

Further problem can be found in the LocalDataSource, when zeroInitialization is set to true. No matter what is set as the initial value of the PV (in BOY), the value will be parsed to java.lang.Double (inside the "initialize(String)" method). A few lines later the initial value on the channel is set to 0 (that is an integer). Integer 0 is not equal to Double 0, and exception will be thrown by the LocalChannelHandler.

The lost events problem seems to be somewhere deep in the pv manager. Open PVManagerPV and put a System.out.println(event) into the pvChanged method of the pvReaderListener. You will see that most of the times the event is [ CONN | VAL ] but sometimes the VAL is missing.

shroffk, 2015-06-02:
I agree, after looking into this a bit the connection problems 2a and 2b seem to be in the pvmanager domain.
I think we should revisit the 2a and 2b after #1174 is resolved.

berryma4, 2015-06-23:
Just a note, there are a lot of comments on the pull request.

berryma4, 2015-06-23:
In the end, i removed the connections part for inside to outside a container.
And, I removed the processVersionDifference for the LinkingContainerModel.

closed 2015-06-30 by berryma4