ControlSystemStudio/cs-studio #1113: Filtering the content of the alarm table based on a (selected) node in the tree

berryma4, 2015-05-08:
- Not implemented yet, up for discussion
  - More of a UI design question 
  - Multiple instances of the alarm table, each showing alarms belonging to a particular node
    - related to filtering

kasemir, 2015-05-08:
I'm for allowing multiple alarm table instances, and for adding a filter to the alarm table.

 * Filter must be very obvious. For example, text at top of alarm table that shows: "Active Filter: /Some Area/System1/Level2"
 * Filtering must not be an accidental side effect of clicking around in the alarm tree. What would work is dragging a node in the alarm tree into the table -> Table now filters on that node and everything below. When using multiple alarm tables, this also makes it obvious which one will update its filter, i.e. the one into which the node is dropped.
 * Can filtering also be configured directly in the alarm table, for example by simply entering the filter path as "/Some Area/System1/Level2"?

utzeln, 2015-05-18:
For now, when that alarm table is split into Current Alarms and Acknowledged Alarms we have the number of alarms / lines shown.

Our initial proposal was to display for instance:
Current Alarms (20 of 56 alarms on /Some Area/System1/Level2)
20 being the number of alarms from the specified node
56 being the total number of alarms from root
 /Some Area/System1/Level2 being the node path used as filter

When grouped, we could have:
Current Alarms (20 of 56) Acknowledged Alarms (13) on /Some Area/System1/Level2

In the View menu, we also could add an option Show -> Show all to reset the filter to the root node:
[image]

This option Show, could be used to specify the path: Show -> Filter.

utzeln, 2015-05-18:
Finally, how to open a new alarm table view?

We could use the view menu and add an option "New Alarm Table"?

jbobnar, 2015-05-18:
To enable drag/drop of alarm tree nodes, the class org.csstudio.alarm.beast.TreeItem has to implement java.io.Serializable and some of the referenced classes need to do that as well. The AlarmClientModel field can be transient, but GDCDataStructure, AADataStructure, and Timestamp should implement Serializable.
The problem is that org.epics.util.time.Timestamp is out of our control (it is part of the PVManager). We can make it transient for now, but it should be fixed eventually. 

Another option is to use the sync and lock options (similar as you can sync the Workspace Navigator and active editor in Eclipse RCP). When sync is on, the table shows whatever is currently selected in the tree. If the table is locked, it will keep the filter that was selected at the time of locking. Here is a demonstration: http://users.cosylab.com/~jbobnar/filter.avi 

Let me know what is the preferred way and if it is DnD, how quick can we get Gabriele to update the org.epics.util jar.

jbobnar, 2015-06-01:
@kasemir @berryma4 
Do you have any thoughts regarding the drag/drop vs sync/lock?

kasemir, 2015-06-01:
There is a use case for only displaying the alarm table. Operators might not care about all the possible alarms, they only need to know the current alarms, so they only open the alarm table.
Consequently, 'sync' is not possible, since there is no alarm tree which provides the current selection.

As for drag/drop, there is no need to transfer the complete tree item (timestamp, guidance, ..). All that the alarm table needs to filter is a path `/config/area/system/subsystem/pv`.
The path of alarm tree items is currently a `String` returned by `getPathName()`. The `AlarmTreePath` offers helpers for splitting/joining such a path.
On the alarm table side, I think it makes sense to allow a plain string/text field for the filter specification, maybe including wildcards. So I could filter on `/area/system` to get only alarms for a subtree as well as `*Vac*` to get any alarm that has Vac(uum) in its path resp. PV name.
To spare users the need to manually enter a correct path, drag & drop from the alarm tree can drop such a string. Technically, this would work by just dragging/dropping the path as a string.

closed 2015-06-09 by kasemir