ControlSystemStudio/cs-studio #1154: Several test projects break/block the build

jbobnar, 2015-05-15:
- org.csstudio.diag.epics.pvtree.test fails because it depends on diag.pvtree version 4.0.0, which does not exist. 
- org.csstudio.remote.test and org.csstudio.servicelocator.test fail because they don't specify the junit framework (instead they import a package) and depend on org.mockito plugin, which does not exist.
- org.csstudio.apputil.ui.test blocks the build because it opens a dialog and does not close it.
- org.csstudio.diag.pvfields.test blocks until forever and also has a test that opens a GUI and waits (the project is also in a wrong plugin folder - it should be in diag and not in appunorganized)

kasemir, 2015-05-27:
Regarding org.csstudio.diag.epics.pvtree.test, the (fragment) MANIFEST.MF asks for a host plugin  org.csstudio.diag.epics.pvtree, minimum version 4.0.0. The current org.csstudio.diag.epics.pvtree is 4.0.1, so it looks OK in the IDE. Where do you see the error?

Regarding org.csstudio.diag.pvfields.test, I'll move it to diag. All but one test in there depend on actual PVs, so those need to be disabled. 

We used to have https://github.com/ControlSystemStudio/cs-studio/wiki/UnitTests as a naming guide:
* Plain JUnit tests: *[[UnitTest]].java
* JUnit Plug-in tests that need the Eclipse runtime but no GUI: *[[HeadlessTest]].java
* JUnit Plug-in tests that need the runtime and the GUI: *[[UiPluginTest]].java
* A demo that requires user input or other interaction. Not a real test, but happens to be implemented with JUnit: *Demo.java

Based on that, the GUIDemo.java was named that way because it opens a GUI and is meant to be run as a manual demo, not an automated test.

What do we now use for source file names so that the code is run as a plain unit test, plugin unit test, or not at all?

jbobnar, 2015-05-28:
> The current org.csstudio.diag.epics.pvtree is 4.0.1, so it looks OK in the IDE. Where do you see the error?

I saw it when building the applications with tycho. It works fine today - maybe an update of p2 repository has fixed it...

willrogers, 2015-07-17:
We have now run into the problem with `org.csstudio.remote.test` and `org.csstudio.servicelocator.test`.  @jbobnar do you know how to fix these builds?  As far as I can tell `org.mockito` does not work with Tycho at present.  We have no building plugins that depend on Mockito.

willrogers, 2015-07-17:
I would be happy to remove these plugins from the POM since they are test plugins and the tests are obviously not being run.

jbobnar, 2015-07-17:
@willrogers We are skipping tests as well. 
For the two plugins you mention the org.mockito bundle is missing. Either the bundle should be installed from somewhere, or it should be changed to org.mockito.mockito-core. That one is provided by maven-osgi-bundles.