ControlSystemStudio/cs-studio #1169: Create a guide for building a standalone product

shroffk, 2015-05-21:
The goal is to build a new cs-studio product using the released modules.

mscheng, 2015-05-22:
good ideal!

shroffk, 2015-06-23:
Started working on a document available here:
https://docs.google.com/document/d/1AVtZ5SvftKIAZOwF3fJuTVF14nQLD5FGUpDgHSIXDwU

shroffk, 2015-06-24:
Following these instructions I was able to build and example product.
Could someone else verify if the instructions work for them too.

shroffk, 2015-06-30:
Once verified I feel this should replace "Chapter 16. Site-Specific Products" in the doc book

MichaelRitzert, 2015-06-30:
I tried to follow the steps and add a few comments in the document, but google docs keeps disconnecting. I will summarize here later. For now I'm stuck at "Step 4". In the wizard, I get the error that "The parent project must have a packaging type of POM". I set the packaging type to "eclipse-feature" as shown in step 2.

shroffk, 2015-06-30:
Argh! my bad...it should be a type pom

MichaelRitzert, 2015-06-30:
Thanks, that did the job. I now get a working .tar. Any chance to start the app directly from the GUI?

My comments:
- Eclipse mars has been released in the meantime. The download link
http://www.eclipse.org/downloads/packages/eclipse-rcp-and-rap-developers/marsrc3
can be changed to
http://www.eclipse.org/downloads/packages/eclipse-rcp-and-rap-developers/marsr
3. Setup the target
- Importing target-dev.target didn't immediately work. The same problem es mentioned in
https://docs.google.com/document/d/1WkbLdEo6j4KGSWabjZsQfelE8ujiDdIemmPca3k_QZM, but the dialog looks different in mars.
I cannot say exactly which of my many tried fixed it in the end.
- "At this stage you should having the following projects in your Package explorer."
=> a refresh might be required
- in step 4, the screenshot of the build.properties edit shows the rootfiles folder that is only created below.

mscheng, 2015-06-30:
Hi,

    I tried to build a product according to the guide and found the same 
problem about eclipse download link. And after importing 
target-dev.target, the problems accurred with "Unkown 
Host:http://mercurialeclipse.eclipselabs.org.codespot.com/hg.wiki/update_site/stable/".

Thanks!
Maosong

shroffk, 2015-07-02:
Based on the discussion during the latest monthly meeting, I will create a product template project and update the documentation accordingly.

shroffk, 2015-07-06:
Based on the feedback here are some instructions with the new template model
https://docs.google.com/document/d/17mL4IJI4rs3MhLCrtleiBJ6RNzdYlozfENLr9ObH09M/edit

jeonghanlee, 2015-07-13:
So, what kind of Eclipse should we use to do this? Mar version for what?

shroffk, 2015-07-13:
I updated the document to use the latest eclipse mars release