ControlSystemStudio/cs-studio #1171: Diirt integration

berryma4, 2015-05-22:
We would like to start using the newest diirt (3.x).

To do:
  - [ ] fork diirt/diirt to controlsystemstudio/diirt
    - remote/diirt -> diirt ; origin/master -> dev ; origin/1.2_cs-studio -> releases
  - [ ] update serviceloader to also use osgi service
  - [ ] configuration / preferences
  - [ ] make sure current Jenkins job works
    - fix p2 publish url and fix documentation
  - [ ] update cs-studio to use new namespaces and api changes
  - [ ] change plugins that used pvmanager extension points to use osgi services (custom formulas and datasources)

shroffk, 2015-06-03:
gabriele said:
```
Hi,

Relooking at the code, the best thing to do is transform the org.diirt.util.Configuration class into an interface. Have a DiirtHomeConfiguration, which is the default. CS-Studio can have a CSStudioCompatibilityConfiguration that takes the information from the current preferences. The product can then look at an eclipse preference “diirtConfigurationFromEclipseProperties”, and at startup change the default configuration.

That would be the way to go. That way is transparent to both the user and the rest of the diirt framework.

We should wait to see if the ITER pans out. If so, I’d do the interface work. It would not be that much.

Gabriele
```

shroffk, 2015-08-26:
Diirt integration issues

workspace projects can’t be used in the IDE - you always have the runtime (out of sync)

refreshing project, updating maven project have no effect.
They are in sync when first loaded into the workspace, but after the first build they are no longer.
Which is due to the fact that the manifest is produced in the build, this manifest timestamp does not match with the one the IDE has and viola you have the out of sync problem.

berryma4, 2015-08-26:
added branch: https://github.com/ControlSystemStudio/cs-studio/tree/diirt_integration

Most of this was taken from various attempts in the past.

migrated to diirt 3.0; 
o.c.u.pvmanager.ui, 
o.c.graphene.opiwidgets, 
o.c.graphene, 
o.c.channel.widgets, 
o.c.diag.epics.pvtree, 
o.c.diag.pvmanager.probe, 
o.c.opibuilder.widgets.extra, 
o.c.u.pvmanager.ui.toolbox, 
o.c.u.pvmanager.widgets, 
o.c.diag.pvmanager.javafx.probe

berryma4, 2015-08-26:
@shroffk are you sure it is the manifest (maven-bundle-plugin) not refreshing the file for the ide?
Or is it another plugin that is messing up the plexus-build-api buildContext that m2e uses:
https://wiki.eclipse.org/M2E_compatible_maven_plugins

willrogers, 2015-10-12:
I've tried the build using the new Diirt p2 repo.  Right now I find two errors in the build.  One I've fixed in #1380, the other is the API change in `org.csstudio.simplepv.pvmanager.test` that I mentioned in #1362.

kasemir, 2015-10-12:
Even with the latest 'master' that mentions the diirt repo in core/pom.xml I get this build error:
````
[INFO] Resolving dependencies of MavenProject: org.csstudio:org.csstudio.utility.pvmanager.ui:3.2.0-SNAPSHOT @ /home/ky9/git/cs-studio/core/utility/utility-plugins/org.csstudio.utility.pvmanager.ui/pom.xml
[INFO] {osgi.os=linux, osgi.ws=gtk, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: org.csstudio.utility.pvmanager.ui 3.2.0.qualifier
[ERROR]   Missing requirement: org.csstudio.utility.pvmanager.ui 3.2.0.qualifier requires 'package org.diirt.vtype 3.0.0' but it could not be found

````

willrogers, 2015-10-12:
Core builds for me with `mvn clean verify -DskipTests=true` (on 2317e4be4d522584e5d7ddbfcaa4ee3744bcbf21).

nickbattam, 2015-10-15:
Since merging this branch the number of checkstyle warnings in the cs-studio code has increased significantly, e.g. from 200 to 1800 issues in Core and 2000 new warnings in Applications.

These are primarily whitespace warnings (e.g. tab characters).

Would it be helpful for me to run though the new files and clean these warnings up? I appreciate this will make any branch merges tricky for a while...

berryma4, 2015-10-29:
@shroffk can we close this?

shroffk, 2015-10-29:
No, I have to complete some changes that were discussed yesterday and today

berryma4, 2015-12-02:
#1470 closed the last part

closed 2015-12-02 by berryma4