ControlSystemStudio/cs-studio #1174: Deadlock between CAJ and pvmanager

shroffk, 2015-05-27:
The problem was discovered when trying to open the datasources view to investigate incorrect connection status messages for pv's inside linking container.
The problem can be reproduced usually by first triggering the false disconnected state in opi screens and then open the datasources view...
 
```
Found one Java-level deadlock:
=============================
"pool-7-thread-1":
  waiting to lock monitor 0x0000000002d3ff58 (object 0x00000000f923caa8, a org.epics.pvmanager.jca.JCAChannelHandler),
  which is held by "PVMgr DataSource Worker 1"
"PVMgr DataSource Worker 1":
  waiting to lock monitor 0x0000000002d3f878 (object 0x00000000f9244a88, a com.cosylab.epics.caj.CAJChannel),
  which is held by "pool-7-thread-1"

Java stack information for the threads listed above:
===================================================
"pool-7-thread-1":
	at org.epics.pvmanager.jca.JCAChannelHandler$3.connectionChanged(JCAChannelHandler.java:404)
	- waiting to lock <0x00000000f923caa8> (a org.epics.pvmanager.jca.JCAChannelHandler)
	at gov.aps.jca.event.DirectEventDispatcher.dispatch(DirectEventDispatcher.java:56)
	at com.cosylab.epics.caj.CAJChannel.setConnectionState(CAJChannel.java:643)
	- locked <0x00000000f923cb40> (a java.util.ArrayList)
	- locked <0x00000000f9244a88> (a com.cosylab.epics.caj.CAJChannel)
	at com.cosylab.epics.caj.CAJChannel.connectionCompleted(CAJChannel.java:353)
	- locked <0x00000000f9244a88> (a com.cosylab.epics.caj.CAJChannel)
	at com.cosylab.epics.caj.impl.handlers.CreateChannelResponse.internalHandleResponse(CreateChannelResponse.java:48)
	at com.cosylab.epics.caj.impl.handlers.AbstractCAResponseHandler.handleResponse(AbstractCAResponseHandler.java:110)
	at com.cosylab.epics.caj.impl.CAResponseHandler.handleResponse(CAResponseHandler.java:139)
	at com.cosylab.epics.caj.impl.CATransport.processRead(CATransport.java:509)
	at com.cosylab.epics.caj.impl.CATransport.processRead(CATransport.java:421)
	at com.cosylab.epics.caj.impl.CATransport.handleEvent(CATransport.java:359)
	at com.cosylab.epics.caj.impl.reactor.lf.LeaderFollowersHandler.handleEvent(LeaderFollowersHandler.java:77)
	at com.cosylab.epics.caj.impl.reactor.Reactor.processInternal(Reactor.java:400)
	at com.cosylab.epics.caj.impl.reactor.Reactor.process(Reactor.java:284)
	at com.cosylab.epics.caj.impl.reactor.lf.LeaderFollowersHandler.run(LeaderFollowersHandler.java:91)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
"PVMgr DataSource Worker 1":
	at com.cosylab.epics.caj.CAJChannel.getConnectionState(CAJChannel.java:748)
	- waiting to lock <0x00000000f9244a88> (a com.cosylab.epics.caj.CAJChannel)
	at org.epics.pvmanager.jca.JCAChannelHandler.disconnect(JCAChannelHandler.java:556)
	- locked <0x00000000f923caa8> (a org.epics.pvmanager.jca.JCAChannelHandler)
	at org.epics.pvmanager.MultiplexedChannelHandler.guardedDisconnect(MultiplexedChannelHandler.java:323)
	at org.epics.pvmanager.MultiplexedChannelHandler.removeWrite(MultiplexedChannelHandler.java:279)
	- locked <0x00000000f923caa8> (a org.epics.pvmanager.jca.JCAChannelHandler)
	at org.epics.pvmanager.DataSource$4.run(DataSource.java:339)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

Found 1 deadlock.
```

shroffk, 2015-06-02:
@msekoranja do you have any idea about this? 
This is not particularly related to linking containers, it seems like pvmanager and CAJ are both trying to update the connectionState for a channel. For what I see it seems like pvmanager is giving up on channels and trying to dispose of them just as CAJ is completing creating the connection.

shroffk, 2015-06-04:
@berryma4 could you please fork the repo from stash to ControlSystemStudio

msekoranja, 2015-06-04:
This is a deadlock because of lock ordering; CAJChannel vs. JCAChannelHandler.
This can be solved in: JCAChannelHandler.
Or in CAJ (if possible) by not holding a lock while calling connectionChanged callback - this should be fixed.

shroffk, 2015-06-04:
Is this already fixed in diirt 3 or CAJ 1.1.15?

berryma4, 2015-06-05:
I can reproduce with a formula:
=tableOf(column("time",subArray('${IOC}:ACQ1:TimeBase.AVAL',0,'${IOC}${DEVICE}:MNbrSamples')),column("Y",subArray('${IOC}:ACQ1:Data',0,'${IOC}${DEVICE}:MNbrSamples')))

shroffk, 2015-06-09:
I have committed a potential fix for issue #1174, the updated handler tries to follow the limitations defined by the direct even dispatcher (of not accessing CAJChannel on the listener thread) and moves the connectionEvent processing onto another thread.
The Pvmanager unit testing pass.
I have also tried to preserve the synchronization around the JCAHandler to ensure that there are no concurrency problems when processing a connection change event.

shroffk, 2015-06-09:
While the deadlock has been resolved we still have a race condition - (which usually becomes apparent when there are hundreds of pv's being used in a linking container ) - if the connect -> disconnect -> connect events are not processed in that order the pv might appear disconnected when they should actually be connected.

berryma4, 2015-06-12:
I've pushed the pvmanager changes to maven.controlsystemstudio.org

closed 2015-06-30 by berryma4

kasemir, 2015-06-30:
Is there a way to tell from a binary product if it includes the updated pvmanager from  maven.controlsystemstudio.org? Version number of the plugin jar? Size in bytes?

berryma4, 2015-06-30:
the version is:
org.csstudio.utility.pvmanager.epics_4.1.0.20150619085355.jar
But, good point, we probably should somehow put the pvmanager version in
there.
ie. 4.1.0.${pvmanager-version}-${date}
I would normally just use the pvmanager version, but the past wrapper went
with cs-studio versions.  This can be changed when we switch to "Diirt".