ControlSystemStudio/cs-studio #1176: Problems with o.c.opibuilder.converter structure

willrogers, 2015-05-27:
The plugin org.csstudio.opibuilder.converter has several problems with the way it's laid out.

* Until recently `converter.jar` was in the repo
** It builds `converter.jar` separately unlike other plugin builds
* `edm2xml.jar` is still in the repo.  I don't know why.
* It uses log4j while the rest of CS-Studio uses java.util.logging
 * Not only that, it bundles an ancient version of the jar in the repo (`log4j-1.2.14.jar`)

It makes sense to me that we would want a standalone jar as output from this build.  I expect we should use the maven shade plugin to bundle in dependencies, although on reflection if we get rid of log4j I don't think we will have any.

Does anyone have an objection to me trying to clean this all up?  Are there some things I don't know about this?

kasemir, 2015-05-27:
I remember seeing those *jar files when I moved the opibuilder plugins and features into applications/opibuilder/* subdirs.
Yes, it would be better to remove them from the repo, same for the log4j jar and its use.

willrogers, 2015-05-27:
There's still something a bit strange with the build which I will come back to next week.

willrogers, 2015-06-04:
When this plugin is built it creates 'converter.jar', which is then bundled inside the plugin jar.

I can think of reasons you might want to do this; in particular it would be nice if you could generate a standalone jar.  There is a `.fatjar` file that is left over from a previous attempt at this. However, I was incorrect in thinking that the converter did not need dependencies, and converter.jar is not a standalone jar, so I don't see any value.  I wanted to make it so that the plugin jar (the one you build anyway) would work as a standalone jar, but the tools you'd use to do it (Maven, Tycho, the maven-shade plugin) don't work well enough together to do it.

Instead I have simplified the build to make the plugin like all the others in https://github.com/ControlSystemStudio/cs-studio/commit/abeff9f3704730c22e2a81726adf3d59d17ea5b2 without any apparent disadvantages.

If this is OK I will submit a pull request.

kasemir, 2015-06-04:
Since the converter.jar as you found it isn't standalone, I agree with removing it as well as the related `.fatjar` etc.

Still, I guess you do want to have a standalone jar for batch conversions, right?

I used to have a similar issue with the scan client, when there was no plain python client lib. I wanted to build a standalone scan_client.jar for use outside of BOY, by jython command line scripts. For that, I created an ant `build.xml`. It happened to be in the plugin directory, but it wasn't used by the normal headless PDE build nor the maven build. I simply invoked it separately from Hudson as a plain ant job.

willrogers, 2015-08-28:
OK.  I think I've found a way to do this, inspired by your suggestion.

In order to simplify this, I have removed dependencies on `org.csstudio.opibuilder`.  In most cases these are `ErrorHandlingUtil` and `ConsoleService`, which can fairly simply be replaced with a call to logging.  The `ui` package has dependencies on Eclipse but I'm not building it with Ant.

I've added a maven profile `-Dbuild.converter.jar`.  If present, it calls the maven antrun plugin to build the jar.  To simplify dependency management, it also directly downloads the one required dependency to make it available to ant.

In the end it all seems to work fine.  I'll test it a bit here before submitting the pull request.

willrogers, 2015-09-01:
Happy to discuss any of these changes.

closed 2015-09-01 by berryma4