ControlSystemStudio/cs-studio #1190: Release 4.1.x (master)

berryma4, 2015-06-09:
Proposing to freeze the master branch to 4.1.x.

  - [ ] resolve caj/pvmanager issue #1174
  - [ ] report BOY regression tests (ControlSystemStudio/opi-regression-tests)
     https://docs.google.com/forms/d/1cKILK51oWpVcy5iDCe4gPK6mvCSvlMWIS5JazuO2_Lc/viewform
  - [ ] make 4.1.x branch
  - [ ] make new github milestone 4.2.0 - testing (master)
  - [ ] make new github milestone 4.1.1 - stable (4.1.x branch)
  - [ ] make / run Jenkins csstudio.product release

Site Representatives, are there any other outstanding issues that need to be addressed before release of 4.1.x?
@berryma4, @kasemir, @shroffk, @richardfearn, @willrogers, @utzeln, @mjclarke01, @jpenning, @ksdj

jpenning, 2015-06-09:
We're fine with this, keep up the good work :-)

Jörg.

kasemir, 2015-06-09:
Yes. Any chance to actually resolve the caj/pvmanager issue?

shroffk, 2015-06-09:
I am still working on this.

willrogers, 2015-06-10:
Fine for me.  Thanks!

mjclarke01, 2015-06-10:
Fine with us.
Thanks,
Matt

shroffk, 2015-06-10:
For Issue #1174 I have submitted a fix, this prevents the deadlock. However that are other issues with this solution which are exacerbated with by the way the current linking container handles connections.
Should this be marked off?

berryma4, 2015-06-24:
The caj/pvamanager issue is "relieved" now that the linking container was reverted to a similar past state, of not trying to to keep connections (widget connections, not pvmanager connections) between widgets inside to outside of linking containers.

berryma4, 2015-06-24:
next step is to report the boy validation tests and branch
(then run product release)

berryma4, 2015-06-25:
https://docs.google.com/forms/d/1cKILK51oWpVcy5iDCe4gPK6mvCSvlMWIS5JazuO2_Lc/viewanalytics

| Test       | Result           |
| ------------- |:-------------:|
| timestamp | 6/25/2015 13:54:39 |
| issue ticket | https://github.com/ControlSystemStudio/cs-studio/issues/1190 |
| Tester | Berryman |
| __BOY Examples__ |
| All BOY Examples | Pass |
| __Issues__ |
| Issue 1081 | Pass |
| Issue 1103 | Pass |
| Issue 1197 | Pass |
| __Performance Tests__ |
| graph_performance_test.opi  | Pass |
| large_led_performace_test.opi | Pass |
| small_led_performace_test.opi | Pass |
| tank_performace_test.opi | Pass |
| test_update_performance_test.opi.opi | Pass |
| l__Rules Tests__ |
| General_rule.opi | Pass |
| Graphs_rule.opi | Pass |
| LED_color_rule.opi | Pass |
| Label_rule.opi | Pass |
| Switch_rule.opi | Pass |
| __WidgetTests__ |
| ActionButton_test.opi | Pass |
| BubbleGraph_test.opi | Pass |
| ComboBox_test.opi | Pass |
| Container_test.opi | Pass |
| Gauge_Meter_Thermometer_Tank_test.opi | Pass |
| LED_test.opi | Pass |
| TextInputWidget_test.opi | Pass |
| TextUpdate_test.opi | Pass |
| XYGraph_test.opi | Pass |

berryma4, 2015-06-25:
Closing ticket, because the punch list is done.  If any new issues come up, report under 4.1.x and regression tests should be added.

Thank you everyone!

closed 2015-06-25 by berryma4