ControlSystemStudio/cs-studio #1197: upgrading to jython 2.7 has significant impact on script loading times

shroffk, 2015-06-10:


berryma4, 2015-06-10:
http://www.jython.org/latest.html#jython-2-7b3-release-notes
By default, site module is imported when using PythonInterpreter Use -Dpython.import.site=false (as of RC1) to not import site at all

shroffk, 2015-06-10:
Reverting back to 2.5.3 resolves this porblme but this requires code changes since the PythonInterpreter has changed and no longer has the close() method

kasemir, 2015-06-10:
Reverting back is not an option for us because that Jython version is out of date.
For example, it cannot handle catching exceptions "e as …", so the PyScanClient won't work.

kasemir, 2015-06-10:
-Dpython.import.site=false makes no difference when I try that here. I'm afraid your example screen nicely shows why scripts should be limited to exceptional cases, can't be used for each widget all the time. Also shows why a significant update in that area is needed, since that script loading currently blocks the whole UI.

berryma4, 2015-06-10:
Odd the VM argument made a difference for me

shroffk, 2015-06-10:
I completely agree...I already spoke to my users and showed them alternative ways of doing this.

But this is a regression, usually my users don't take kindly to being told that a lot of their old screens are now broken :)

I still feel at times formulas can be very useful (like in this case they don't want to create IOCs to control the color)

kasemir, 2015-06-11:
I ran the current CSS with your Widgets_with_python_scripts.opi in JProfiler, and it's indeed spending all the time in org.python.util.PythonInterpreter <init>. I get the same ~100 seconds as Kunal, even with -Dpython.import.site=false. I don't have the source code for org.python.jython_2.7.0.b3.jar, so can't look further inside what exactly is happening there.

What helps: The previous org.python_2.5.2.20140409 plugin was expanded, while org.python.jython_2.7.0.b3.jar is a jar file. I've unpacked it in the product's plugin dir, then updated configuration/org.eclipse.equinox.simpleconfigurator/bundles.info. Now the example loads in ~35 seconds.

So we need to package that python plugin in expanded form, as a directory, not jar.

kasemir, 2015-06-11:
{{{
cd plugins
mkdir org.python.jython_2.7.0.b3
cd org.python.jython_2.7.0.b3
unzip ../org.python.jython_2.7.0.b3.jar
rm org.python.jython_2.7.0.b3.jar
}}}

berryma4, 2015-06-11:
Just added Eclipse-BundleShape:dir
and added source, also upgraded to 2.7.0 release.
ControlSystemStudio/maven-osgi-bundles@4b85167c32c1d082f01d891d64fde4f481251f42

berryma4, 2015-06-11:
I added a singleton for jython initialization, this seems to speed things up

berryma4, 2015-06-11:
On branch: issue_1197_jython_slow
@c5a055793a0b4df480cbbe7fdfcb8723ea3f18d0

berryma4, 2015-06-11:
new error in /BOY Examples/5_4_Python_Script.opi :
```java
2015-06-11 12:15:01 ERROR: Failed to register scripts/KnobValueDialog.py. 
SyntaxError: ('encoding declaration in Unicode string', ('<script>', 0, 0, ''))

2015-06-11 12:15:01 ERROR: Failed to register scripts/ChangePosAndSize.py. 
SyntaxError: ('encoding declaration in Unicode string', ('<script>', 0, 0, ''))
```

kasemir, 2015-06-11:
I don't think the singleton is a good idea.
Scripts can run for a longer time when they spawn threads, and the singleton would then replace the "widget" etc. of an already running script instance.

Is that new error in /BOY Examples/5_4_Python_Script.opi also only happening with the singleton? Or is that after the change from b3 to release?

shroffk, 2015-06-11:
I do not see the errors with the BOY examples using jython 2.7.0b3

berryma4, 2015-06-11:
Yeah, the singleton branch kills it.

berryma4, 2015-06-11:
On branch: issue_1197_jython_slow
@ad33f16f40998c16b33e36dbb5b09276d90eb4e1
Okay, I switched from a singleton to "PythonInterpreter.threadLocalStateInterpreter(dict);"
This allows setlocals.
I also removed the PythonInterpreter that extended PythonInterpreter.

closed 2015-06-12 by berryma4

berryma4, 2015-06-12:
#1190

kasemir, 2015-07-06:
While the shared interpreter is good for performance, I now need to restart CSS after changing scripts.
Simple closing/reopening the *.opi that uses a script no longer suffices.

Not a bug, because the runtime performance is right now more important.
Just mentioning it here in case others run into the same quirk.