ControlSystemStudio/cs-studio #1205: boy: symbol widgets and svgs

berryma4, 2015-06-15:
I get the following error when I try to select an svg image for a multi state symbol.
[image]

@jbobnar do you know about this?

``` java
An error has occurred. See error log for more details.
loader constraint violation in interface itable initialization: when resolving method "org.apache.batik.dom.AbstractDocument.createEvent(Ljava/lang/String;)Lorg/w3c/dom/events/Event;" the class loader (instance of org/eclipse/osgi/internal/loader/EquinoxClassLoader) of the current class, org/apache/batik/dom/svg/SVGOMDocument, and the class loader (instance of <bootloader>) for interface org/w3c/dom/events/DocumentEvent have different Class objects for the type org/w3c/dom/events/Event used in the signature
```

jbobnar, 2015-06-15:
Does your product include this fragment: https://github.com/ControlSystemStudio/cs-studio/tree/master/core/platform/platform-plugins/org.csstudio.jre6.fragment?

It's a problem with eclipse trying to be backward compatible with java 1.5, which did not provide the org.w3c.dom.events package, so eclipse tries to load those classes from org.w3c.dom.events plugin. Java 1.6 and later do provide that package within JRE, so you easily end up with two versions of the same class.

berryma4, 2015-06-15:
@utzeln I forgot, Jaka is on vacation.  Do you also see this issue on master?

berryma4, 2015-06-15:
Maybe you're not on vacation ....
So, this fragment doesn't seem happy in the IDE, but it gets picked and works fine with tycho.

jbobnar, 2015-06-15:
I am on vacation...
 
You need that fragment in your product definition. It tells eclipse to stop looking for the org.w3c plugin and just take the code from jre. If you do include that fragment and are still getting this error, than make sure that you do not explicitly use org.w3c.dom.events plugin anywhere.

closed 2015-06-24 by berryma4