ControlSystemStudio/cs-studio #1210: rebuilding org.csstudio.sns and org.csstudio.product based on cs-studio-4.1.0

mscheng, 2015-06-23:
I rebuild org.csstudio.sns and get following errors when I test cs-studio-4.1.0. The similar errors show when I rebuild org.csstudio.product.
[INFO] Computing target platform for MavenProject: org.csstudio:org.csstudio.sns.feature.eclipse:4.1.0-SNAPSHOT @ /home/mscheng/gitcssrepository/org.csstudio.sns/features/org.csstudio.sns.feature.eclipse/pom.xml
[INFO] Resolving dependencies of MavenProject: org.csstudio:org.csstudio.sns.feature.eclipse:4.1.0-SNAPSHOT @ /home/mscheng/gitcssrepository/org.csstudio.sns/features/org.csstudio.sns.feature.eclipse/pom.xml
[INFO] {osgi.os=linux, osgi.ws=gtk, org.eclipse.update.install.features=true, osgi.arch=x86}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: org.csstudio.sns.feature.eclipse.feature.group 4.1.0.qualifier
[ERROR]   Missing requirement: org.eclipse.equinox.simpleconfigurator.manipulator 2.0.0.v20131217-1203 requires '[ava.package org.eclipse.equinox.frameworkadmin [2.0.0,3.0.0)' but it could not be found
[ERROR]   Cannot satisfy dependency: org.csstudio.sns.feature.eclipse.feature.group 4.1.0.qualifier depends on: org.eclipse.equinox.p2.user.ui.feature.group 0.0.0
[ERROR]   Cannot satisfy dependency: org.eclipse.equinox.p2.core.feature.feature.group 1.3.0.v20140523-0116 depends on: org.eclipse.equinox.simpleconfigurator.manipulator [2.0.0.v20131217-1203]
[ERROR]   Cannot satisfy dependency: org.eclipse.equinox.p2.user.ui.feature.group 2.2.0.v20140523-0116 depends on: org.eclipse.equinox.p2.core.feature.feature.group [1.3.0.v20140523-0116]
[ERROR]   Cannot satisfy dependency: org.eclipse.equinox.p2.user.ui.feature.group 2.2.0.v20140911-0143 depends on: org.eclipse.equinox.p2.core.feature.feature.group [1.3.0.v20140523-0116]
[ERROR]   Cannot satisfy dependency: org.eclipse.equinox.p2.user.ui.feature.group 2.2.0.v20150129-0838 depends on: org.eclipse.equinox.p2.core.feature.feature.group [1.3.0.v20140523-0116]

kasemir, 2015-06-23:
How are you building it?

I'm using maven settings.xml with <csstudio.composite.repo> that point to a local composite repo, which in turn list the output of other locally built repos, i.e.

maven-osgi-bundles/repository/target/repository,
cs-studio-thirdparty/repository/target/repository,
cs-studio/core/p2repo and 
cs-studio/applications/p2repo

It built OK last night.

mscheng, 2015-06-24:
Hi, kasemir,
I build it using the same method and settrings as your.

Before, I build cs-studio/applications, I get following errors:
[INFO] Resolving dependencies of MavenProject: org.csstudio:org.csstudio.email.test:1.0.0-SNAPSHOT @ /home/mscheng/gitcssrepository/cs-studio/applications/appunorganized/appunorganized-plugins/org.csstudio.email.test/pom.xml
[INFO] {osgi.os=linux, osgi.ws=gtk, org.eclipse.update.install.features=true, osgi.arch=x86}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: org.csstudio.email.test 1.0.0.qualifier
[ERROR]   Missing requirement: org.eclipse.equinox.simpleconfigurator.manipulator 2.0.0.v20131217-1203 requires '[ava.package org.eclipse.equinox.frameworkadmin [2.0.0,3.0.0)' but it could not be found
[ERROR]   Cannot satisfy dependency: org.csstudio.email.test 1.0.0.qualifier depends on: bundle org.csstudio.testsuite 1.0.0
[ERROR]   Cannot satisfy dependency: org.csstudio.testsuite 1.0.0.qualifier depends on: bundle org.eclipse.jdt.junit.core 3.6.1
[ERROR]   Cannot satisfy dependency: org.eclipse.jdt.junit.core 3.7.300.v20140409-1618 depends on: bundle org.eclipse.equinox.simpleconfigurator.manipulator [2.0.0,3.0.0)

So, I disable  org.csstudio.email.test and org.csstudio.testsuite modules, and then cs-studio/applications build successfully.

But, I get above errors when i rebuild org.csstudio.sns and org.csstudio.product.

kasemir, 2015-06-24:
My .m2/settings.xml has
```
       <!-- Skip unit tests -->
        <maven.test.skip>true</maven.test.skip>
        <skipTests>true</skipTests>
```
It still compiles email.test, but doesn't try to execute it:
```
INFO] Computing target platform for MavenProject: org.csstudio:org.csstudio.email.test:1.0.0-SNAPSHOT @ /usr/local/hudson/config/jobs/CSS_3_cs-studio/workspace/applications/appunorganized/appunorganized-plugins/org.csstudio.email.test/pom.xml
[INFO] Resolving dependencies of MavenProject: org.csstudio:org.csstudio.email.test:1.0.0-SNAPSHOT @ /usr/local/hudson/config/jobs/CSS_3_cs-studio/workspace/applications/appunorganized/appunorganized-plugins/org.csstudio.email.test/pom.xml
…
[INFO] org.csstudio.email.test ............................ SUCCESS [  0.330 s]
```

I don't see any mention of org.eclipse.equinox.simpleconfigurator.manipulator in my logs.

kasemir, 2015-06-24:
Just to make sure: I'm building off the 'master' branch.

mscheng, 2015-06-24:
Hi, kasemir
     I solve it by removing all contents in .m2/repository and local repo, and rebuild maven-osgi-bundles, cs-studio-thirdparty, cs-studio, org.csstudio.product and org.csstudio.sns successfully.
    The problem caused by network(internet) issues. Some JAR files couldn't be download correctly and copied to corresponding directory, which resulted in missing requirement issues. So, everyone makes sure your internet speed is smooth when you build cs-studio.
kasemir, many thanks!

closed 2015-06-24 by mscheng