ControlSystemStudio/cs-studio #1214: Linking container error when missing the linked opi_file

kasemir, 2015-06-24:
As long as the <opi_file> target of a linking container widget is accessible, all is fine.
But if the target doesn't exist,  both the display editor and runtime refuse to load at all with an error "Cannot open /that/missing/embedded.opi".

Since this includes the editor, there is no way to for example enter a different opi_file target. Have to use a text editor for this, except even "Open With/Text Editor" is no longer available from the Navigator context menu unless I restart CSS.

When creating a new *.opi file with a new LinkingContainer, where the opi_file target is initially empty until one enters a valid *.opi file path, there is an error message "java.lang.Exception: Cannot open  (No such file or directory)" for the empty opi_file target.

berryma4, 2015-06-24:
... right when I thought I could take a break from the linking container.

kasemir, 2015-06-24:
Hi Eric:

Yes, this almost starts to suck.
I've fixed the easy part, the initially empty opi_file in the editor.

kasemir, 2015-06-24:
Branch bug1214

kasemir, 2015-06-24:
I think I fixed it. Now get a logged warning instead of the crashes.

closed 2015-06-24 by berryma4

kasemir, 2015-06-24:
Thanks for getting it onto the 4.1 branch.