ControlSystemStudio/cs-studio #1221: Build error because of missing org.csstudio.simplepv

kasemir, 2015-06-30:
A build of 'applications' fails with
```
[ERROR]   Missing requirement: org.csstudio.opibuilder 4.0.103.qualifier requires 'bundle org.csstudio.simplepv 1.0.0' but it could not be found
```

core/unorganized/unorganized-plugins/org.csstudio.simplepv is built, but it is not placed in unorganized/unorganized-repository/target/repository/plugins/ because the unorganized/unorganized-repository/category.xml does not include any feature which would cover org.csstudio.simplepv

--> Add a simplepv feature?

kasemir, 2015-06-30:
@Kunal @berryma4  Adding a simplepv feature to core/unorganized would be the easiest thing to do right now, smallest number of changes.

On the other hand, simplepv is only used by opibuilder, so moving core/unorg/*simplepv* and apps/unorg/*simplepv* into opibuilder would consolidate it all in one place.

Preference?

berryma4, 2015-06-30:
I would vote to add them to opibuilder, so it's not mistaken as a core api.

kasemir, 2015-06-30:
Yes, "not mistaken as a core api" is a good reason for moving it to / hiding it in the opibuilder.

I'll do that.

berryma4, 2015-07-01:
Added to master and 4.1.x

PR #1222 (2015-07-01): SimplePV: Move into opibuilder
org.csstudio.simplepv and org.csstudio.simplepv.pvmanager used to be under core/unorganized,
org.csstudio.simplepv.vtypepv used to be under applications/appunorganized.

Now they're under the opibuilder, since nothing else uses them.

org.csstudio.simplepv.utilitypv has been deleted.

Fixes #1221
closed 2015-07-01 by berryma4