ControlSystemStudio/cs-studio #1234: Minimise CS-Studio to tray icon

willrogers, 2015-07-08:
It's possible to intercept clicking the cross in the top-right of the CSS window and minimise CSS to a tray icon.  You can then click the tray icon to restore or exit CSS.  We discussed this briefly at the code-a-thon.

I've got a commit to do this, although there is some subtlety with WorkbenchWindowAdvisors that I'm not sure I understand.

kasemir, 2015-07-09:
Can you please elaborate for me who forgot details, and those not at the code-a-thon?
Is the point of this to intercept the 'close window', so instead of closing CSS it's minimized?

willrogers, 2015-07-10:
Yes, sorry.

I think that this would be an option in preferences.  If selected:

* Clicking the top-right little cross makes CSS go to a little icon in the tray
* Clicking the tray icon makes the CSS window come back
* Right-clicking the tray icon gives options 'Open' or 'Exit'

This is potentially useful if you are opening standalone OPI screens and you don't need the CSS window right now.  There could also be a 'Start minimised' option

There are a couple of tricky points:

* How do you handle more than one main CSS window?
* What happens if CSS is minimised and you (for example) right-click and 'Open in databrowser'?

Spotify is an example of a program that has this type of behaviour.

I could work on this if enough people think that it is a useful feature.

closed 2021-04-14 by willrogers