ControlSystemStudio/cs-studio #1236: Deadlock in alarm client

shroffk, 2015-07-09:
With the new release we have stumbled upon another deadlock issue in the alarm client. It happened when from the alarm tree the user is configuring an AlarmTreeItem and an update is received from the jms thread.
 ```
Found one Java-level deadlock:
=============================
"ActiveMQ Session Task":
  waiting to lock monitor 0x0000000002cbd7b8 (object 0x00000000d7a630f0, a org.csstudio.alarm.beast.client.AlarmTreeItem),
  which is held by "Worker-6"
"Worker-6":
  waiting to lock monitor 0x0000000003a391d8 (object 0x00000000d7b44670, a org.csstudio.alarm.beast.client.AlarmTreePV),
  which is held by "ActiveMQ Session Task"

Java stack information for the threads listed above:
===================================================
"ActiveMQ Session Task":
	at org.csstudio.alarm.beast.client.AlarmTreeItem.maximizeSeverity(AlarmTreeItem.java:330)
	- waiting to lock <0x00000000d7a630f0> (a org.csstudio.alarm.beast.client.AlarmTreeItem)
	at org.csstudio.alarm.beast.client.AlarmTreeItem.setAlarmState(AlarmTreeItem.java:291)
	- eliminated <0x00000000d7b44670> (a org.csstudio.alarm.beast.client.AlarmTreePV)
	at org.csstudio.alarm.beast.client.AlarmTreeLeaf.setAlarmState(AlarmTreeLeaf.java:111)
	- eliminated <0x00000000d7b44670> (a org.csstudio.alarm.beast.client.AlarmTreePV)
	at org.csstudio.alarm.beast.client.AlarmTreePV.setAlarmState(AlarmTreePV.java:216)
	- locked <0x00000000d7b44670> (a org.csstudio.alarm.beast.client.AlarmTreePV)
	- locked <0x00000000d7a2f180> (a org.csstudio.alarm.beast.ui.clientmodel.AlarmClientModelRoot)
	at org.csstudio.alarm.beast.ui.clientmodel.AlarmClientModel.updatePV(AlarmClientModel.java:987)
	at org.csstudio.alarm.beast.ui.clientmodel.AlarmClientCommunicator$UpdateAction.run(AlarmClientCommunicator.java:68)
	at org.csstudio.alarm.beast.ui.clientmodel.AlarmClientCommunicator.handleMapMessage(AlarmClientCommunicator.java:423)
	at org.csstudio.alarm.beast.ui.clientmodel.AlarmClientCommunicator.access$1(AlarmClientCommunicator.java:354)
	at org.csstudio.alarm.beast.ui.clientmodel.AlarmClientCommunicator$2.onMessage(AlarmClientCommunicator.java:172)
	at org.apache.activemq.ActiveMQMessageConsumer.dispatch(ActiveMQMessageConsumer.java:1229)
	- locked <0x00000000c6c472f8> (a java.lang.Object)
	at org.apache.activemq.ActiveMQSessionExecutor.dispatch(ActiveMQSessionExecutor.java:134)
	at org.apache.activemq.ActiveMQSessionExecutor.iterate(ActiveMQSessionExecutor.java:205)
	at org.apache.activemq.thread.PooledTaskRunner.runTask(PooledTaskRunner.java:122)
	at org.apache.activemq.thread.PooledTaskRunner$1.run(PooledTaskRunner.java:43)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
"Worker-6":
	at org.csstudio.alarm.beast.client.AlarmTreePV.getCurrentSeverity(AlarmTreePV.java:83)
	- waiting to lock <0x00000000d7b44670> (a org.csstudio.alarm.beast.client.AlarmTreePV)
	at org.csstudio.alarm.beast.client.AlarmTreeItem.maximizeSeverity(AlarmTreeItem.java:342)
	- locked <0x00000000d7a630f0> (a org.csstudio.alarm.beast.client.AlarmTreeItem)
	at org.csstudio.alarm.beast.ui.clientmodel.AlarmClientModel.updateEnablement(AlarmClientModel.java:966)
	at org.csstudio.alarm.beast.ui.clientmodel.AlarmClientCommunicator$EnableAction.run(AlarmClientCommunicator.java:93)
	at org.csstudio.alarm.beast.WorkQueue.performQueuedCommands(WorkQueue.java:141)
	at org.csstudio.alarm.beast.ui.clientmodel.AlarmClientCommunicator.setQueueMode(AlarmClientCommunicator.java:212)
	at org.csstudio.alarm.beast.ui.clientmodel.AlarmClientModel.readConfiguration(AlarmClientModel.java:505)
	at org.csstudio.alarm.beast.ui.clientmodel.ReadConfigJob.run(ReadConfigJob.java:49)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:54)

Found 1 deadlock.
```

kasemir, 2015-07-09:
Good catch, will have to look at it.

kasemir, 2015-07-14:
The AlarmClientModel javadoc describes the lock hierarchy: ".. lock affected items from the root down."

In this deadlock, AlarmClientCommunicator (ActiveMQ Session Task) received an update, locked the AlarmClientModelRoot followed by an AlarmTreePV, then tries to lock another AlarmTreeItem, which should be OK because it holds the AlarmClientModelRoot.

Concurrently, the AlarmClientModel (Worker-6) applies JMS updates received while loading config from RDB. It calls AlarmTreeItem.maximizeSeverity(), which traverses up to the parent, i.e. modifies several PVs, but failed to lock the AlarmClientModelRoot.
AlarmClientModel.updateEnablement() actually held the root lock, but released it before calling maximizeSeverity().

Looks like I didn't follow my own Javadoc in AlarmClientModelRoot. Will fix that and check other calls to maximizeSeverity() for compliance.

kasemir, 2015-07-14:
For what it's worth, I don't think this is a new bug. Was like this since Feb. 2013, just went unnoticed because it can only happen while you reconfigure.

PR #1243 (2015-07-14): Bug 1236 alarmclient deadlock
Fixes #1236:

maximizeSeverity() needs to be called with locked 'root', because it traverses the alarm tree.

Replaced several sync'ed sections with volatile variables and safe copies.
commit 94ad032e (2015-07-14): alarm client: Lock root when maximizing severity up to root

Fixes #1236
commit 992ef011 (2015-07-14): alarm client: Lock root when maximizing severity up to root

Fixes #1236
closed 2015-07-14 by berryma4