ControlSystemStudio/cs-studio #1237: Jython Script problems

kasemir, 2015-07-10:
The changes to fix Jython memory leaks and re-use the Jython interpreter/state introduced two new problems.

```
java.lang.NullPointerException
        at org.csstudio.opibuilder.widgetActions.ExecutePythonScriptAction.dispose(ExecutePythonScriptAction.java:169)
        at org.csstudio.opibuilder.editparts.AbstractBaseEditPart.deactivate(AbstractBaseEditPart.java:302)
``
because it's trying to access the member variable `state` which is no longer set.

PyScanClient scripts fail when opening a display within the same CSS instance for the second time because something inside the url2lib or socket lib is closed, yet still reused because it's the same Jython interpreter/state: https://github.com/PythonScanClient/PyScanClient/issues/18

closed 2015-07-13 by berryma4

kasemir, 2016-02-23:
@berryma4 @jbobnar
 
Unsure if this is a fault in Jython, or the way we use it, or the Jython version of _socket.py, but here's what's happening in scan related scripts:

When opening a display with a script, BOY `JythonScriptStore` creates a Jython interpreter via 
````
state = Py.getSystemState();
interp = PythonInterpreter.threadLocalStateInterpreter(state.getDict());
````
That interpreter then compiles the script and executes it whenever a trigger PV changes.
As the *.opi is closed, JythonScriptStore calls
````
interp.close();
interp.cleanup();
````

Now take the PyScanClient or another script that uses urllib2 which in turn is based on _socket.py.
The Jython variant of that library, `org.python.jython_2.7.0.release/Lib/_socket.py`, internally creates a thread pool:
````
NIO_GROUP = NioEventLNioEventLoopGroup(..)
..
def _shutdown_threadpool():
    ..
    NIO_GROUP.shutdownGracefully(0, 100, TimeUnit.MILLISECONDS)
    log.debug("Shut down thread pool", extra={"sock": "*"})

sys.registerCloser(_shutdown_threadpool)
````

As JythonScriptStore deletes the Jython interpreter, `_socket._shutdown_threadpool()` shuts the thread pool down. When enabling logging, one can see the "Shut down thread pool" message.

That's all fine until opening the same *.opi file again.
JythonScriptStore calls `PythonInterpreter.threadLocalStateInterpreter()`, compiles the script again, but this time it doesn't recompile urllib2 and _socket.py. It uses the previously generated binary, which includes the NIO_GROUP instance that has already been shut down. The urllib2 code thus gets 'connection' errors as a result of a "RejectedExecutionException: event executor terminated".

I've worked around this issue in the PyScanClient by re-creating the thread pool and then also registering for that one to be shut down:
````
import sys, _socket
if _socket.NIO_GROUP.isShutdown():
    print "RE-CREATEING NIO_GROUP"
    _socket.NIO_GROUP = _socket.NioEventLoopGroup(2, _socket.DaemonThreadFactory("PyScan-Netty-Client-%s"))
    sys.registerCloser(_socket._shutdown_threadpool)
````
As far as I can tell, that "works", but it's a hack because I don't think I should peek into the internals of _socket.py.

Jython scripts started from Action Buttons don't seem to have that problem because `ExecutePythonScriptAction` always creates a new state and interpreter:
````
state = new PySystemState();
interpreter = new PythonInterpreter(null,state);
````
It always recompiles all the python code, not recycling a _socket.py binary with shut-down NIO_GROUP.
Works fine, but it takes about 3 seconds(!!) to create a totally new state and interpreter, and with BOY that means the whole display freezes for 3 seconds when you first press the Action Button.

So for now JythonScriptStore is in between: It closes/disposes the interpreter, which invokes at-exit type cleanup routines in for example the _socket.py code. But then it reuses that code to get slightly faster startup the second time you open a display, requiring hacks to recreate some _socket.py initialization code.

berryma4, 2016-02-23:
So, yes I've run into this annoyance recently.  I think the Action Buttons and the scripts being different makes this even worse.  I thought I would make a nice jython based scan client using BOY, but became frustrated that I couldn't share the interp of the button that submitted the scan (and that it takes 3 seconds to submit the scan), and only action buttons can submit scans (because of the mentioned thread pool shutdown).

We really need to agree on the expected behavior ... I'm just afraid of backward compat.

My answer was to make a simple view for a user (not BOY), which I rather not do in the future.

kasemir, 2016-02-23:
The latest PyScanClient includes the _socket.py workaround, so both PV-triggered scripts and action buttons "work".

I agree that changing anything in BOY asks for unexpected side effects. To make it safe, you need several confined jython interpreters. But as long as that's all on the UI thread, we can't afford it.

Going forward, I'm trying to determine what to do in the display builder project.
For now, it has one jython interpreter per *.opi file. When the first script on the *.opi file needs jython, it's created. Takes 3 seconds, but that's in the background, not freezing the UI. All widgets and actions within the same *.opi share the same script interpreter and one single-threaded executor.
No matter if 1 or 100 scripts, there's one initial 3 second delay until the first script runs on a newly opened *.opi.
Scripts within one *.opi can share variables, since it's the same interpreter. Could be a feature.
Only one script per *.opi is executing at a time. If N PVs trigger, the N script invocations queue up until they're executed. They do execute in the background, so OK to have one script "run forever".

So in short the display builder does a full create & dispose of jython for each *.opi, but in the background. For N executing *.opi files, there are N jython instances & threads, compared to BOY with (N + number of action buttons) jython instances which share some state, all executed by 1 UI thread.

jbobnar, 2016-02-24:
Ever since we found those major memory leaks we try to avoid jython in any way possible. ITER also encourages people not to use jython because of slow opi startup (they had a case when replacing the jython with javascript stripped off about 10-15 seconds from the startup time, but I think that was before 2.7). If loading happens in the background and is not too CPU intensive, it might be OK.

> When the first script on the *.opi file needs jython, it's created. 

If the script is loaded in the background, it might be better to create the jython interpreter eagerly instead of lazily. When you load the opi you probably know if there are any jython scripts, and if yes you could create the interpreter. It's more intensive at startup, but the response of the GUI will be better.

> If N PVs trigger, the N script invocations queue up until they're executed. 

Might be useful to have a queue of only 2 (as an option). One task is being executed and one waiting, If another task is offered you replace the one that is waiting. That could be useful sometimes (not always, because it depends what is done in a script).

kasemir, 2016-02-24:
Yes, scripts should certainly be avoided. Basically, all the EDM-type displays should be free of scripts. If every widget has a script (rule..) because instead of "alarm sensitive border" one prefers a script that wiggles the "y" position, that'd be a misuse.
Scripts are for interfacing with an RDB, scan server, online services, things that would be impossible with EDM. You're right that javascript is much faster than Jython. That's a real pity, because on beam lines python has many good uses outside of CSS, and being able to use the same syntax and often also the same existing code in display scripts is nice.

As for eager/lazy, displays that use no scripts will never create a Jython (nor Javascript) interpreter in either case, and again that should be 95% of all displays.
If a *.opi uses a script, the eager/lazy makes little difference in performance, but a big one in modularity.
a) Lazy: Load display, start display runtime, interpreter is created on the first widget property that needs it.
b) Eager: Load display, interpreter is created on the first widget property that refers to a script, start display runtime.

All of this happens in background threads. The main "Load display" time is spent in actually getting the file content. Parsing the XML, then starting the runtime is comparably quick. So eager start of the interpreter only starts it a tiny bit earlier. But in the lazy case, the model has no idea about PVs & scripts. That's totally hidden inside the runtime module. Good separation. In the eager case, the module loader needs to know about scripts, which is really a runtime matter. Worse, you now need two modes of loading a display: For the editor, without creating runtime script interpreters, and for the runtime mode, where you do create them.

As for queue length: Yes, it could be length limited just to avoid running out of memory. With option to replace the latest entry instead of just dropping when the queue is full. Having that configurable, however, would change the behavior of scripts, maybe unknown to the person who at the time writes the scripts. Best would be one, known, fixed setup. Like queue of 100, then replace: Normally, that means all sane scripts get executed. But if you have too many slow scripts, the system self-protects by not growing the queue until it exhausts memory, and you only get the 'latest' script requests to run.

PR #1238 (2015-07-13): opibuilder: Fix ExecutePythonScriptAction NPE and use separate interp.
This reverts back from the threadLocalStateInterpreter to a PythonInterpreter for script actions invoked by action buttons.
The threadLocalStateInterpreter leaves the python PyScanClient/urllib2/socket lib in a state that results in https://github.com/PythonScanClient/PyScanClient/issues/18

This reversal of the PythonInterpreter -> threadLocalStateInterpreter commits only affects script actions invoked by action buttons, not display scripts triggered by PVs.

In addition, it fixes a bug introduced by commit e95684b which t [... cut]