ControlSystemStudio/cs-studio #1241: OPIBuilder rap plugin fails to build

willrogers, 2015-07-14:
Building rap has always been a mystery here at DLS.  Right now it's failing on `master` due to this error.

Is this happening for others?

```
[ERROR] Failed to execute goal org.eclipse.tycho:tycho-compiler-plugin:0.22.0:compile (default-compile) on project org.csstudio.opibuilder.rap: Compilation failure: Compilation failure:
[ERROR] /scratch/will/css/cs-studio/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder.rap/src/org/csstudio/webopi/ApplicationWorkbenchWindowAdvisor.java:[54]
[ERROR] runOther(path);
[ERROR] ^^^^^^^^
[ERROR] The method runOther(IPath) is undefined for the type ApplicationWorkbenchWindowAdvisor
[ERROR] /scratch/will/css/cs-studio/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder.rap/src/org/csstudio/webopi/ApplicationWorkbenchWindowAdvisor.java:[62]
[ERROR] runOther(path);
[ERROR] ^^^^^^^^
[ERROR] The method runOther(IPath) is undefined for the type ApplicationWorkbenchWindowAdvisor
[ERROR] 2 problems (2 errors)
```

chiijung, 2015-07-17:
I have the same error messages. Do you have any idea to fix this problem?

jbobnar, 2015-07-17:
According to file history broken by this commit: 0b853f57633550492c46db098fdab809d83087bd

willrogers, 2015-07-17:
Right, that's probably my fault.  It was a pretty major merge after moving a lot of plugins around.

Thanks!

berryma4, 2015-09-29:
status?

willrogers, 2015-09-29:
I think this is fixed.

closed 2015-09-29 by berryma4