ControlSystemStudio/cs-studio #1250: org.swt.* feature?

berryma4, 2015-07-15:
@kasemir, is there a feature for org.csstudio.swt.* in appunorganized?
Jenkins is failing because of this.

berryma4, 2015-07-15:
More specific, the databrowser module is failing because it can't find org.csstudio.swt.rtplot

kasemir, 2015-07-15:
I wonder how that used to work. As recent as July 13 it built OK for me.
The org.csstudio.trends.databrowser2.feature includes the org.csstudio.swt.rtplot.

kasemir, 2015-07-15:
Ah, it might be OK when using 'mvn verify' in applications, because it finds the org.csstudio.swt.rtplot sources and compiles them because of org.csstudio.trends.databrowser2.feature.

When you compile module by module, it doesn't work:
applications/appunorganized ignores org.csstudio.swt.rtplot because its in no appunorganized-feature,
applications/databrowser then misses it.

So yes, needs some appunorganized-feature like o.c.swt.rtplot.feature.

kasemir, 2015-07-15:
Are you creating one, or should I do that?

berryma4, 2015-07-15:
I will submit a pull request tomorrow.
Thank you!

PR #1255 (2015-07-17): o.c.utilities.feature: added rtplot, so it's visible from p2
resolves #1250
PR #1257 (2015-07-17): o.c.utilities.feature: add rap utilities feature
closes #1250
closed 2015-07-17 by berryma4