ControlSystemStudio/cs-studio #1254: Update org.csstudio.opibuilder to 4.1.0

willrogers, 2015-07-17:
Should we update `org.csstudio.opibuilder` to 4.1.0?  There have been significant changes including an interface change.

Or even 5.0.0?

kasemir, 2015-07-17:
Should certainly increment.

"4.2" would make sense because it'll then match the CSS milestone. The *.opi files are still compatible, so an increment to "5.x" is not necessary for that. But you're right that the behavior of the interface can be quite different when you're _not_ starting out in the OPIRuntime perspective, because the first screen you open will no longer appear in the 'editors' stack.

So call it "5"?

Then, since the operator interface is the most visible part of CSS, do we also call it CSS 5.0? Or do we keep 4.x there because it's based on Eclipse 4.x?

willrogers, 2015-07-20:
I think it probably is a major change to opibuilder.  Whether it warrants moving all of CS-Studio to 5.0 I don't know.

Your call, I think.

berryma4, 2015-10-12:
did we do this?  We should.
I would go with 4.2.
I would save 5.0 for Kay's new version.

willrogers, 2015-10-12:
Happy with whatever the consensus is.

closed 2018-06-12 by berryma4