ControlSystemStudio/cs-studio #1263: Action button disabled if PV read-only

willrogers, 2015-07-21:
We've got a subtle problem with the Action Button.

If the action button is attached to a PV, and the PV is read-only, the action button is disabled.  This would be fine if the button only *writes* the PV, but in fact the action could do any number of things.  In our case the PV is used for button alarm status and the action opens another OPI.

The simplest way I can see to get around this is a preference 'Disable if PV read-only' on the action button.  It is another configuration option though.

Alternatively, we could never disable the action button (in few cases the action is to write that PV).

Possible workarounds:

* We use a rule to determine our alarm status
* We create an 'Enabled Action Button' that doesn't disable

@kasemir @gilesknap

kasemir, 2015-07-21:
Well, the 'normal' behavior looks good to me. If the button has a PV, that's usually for the purpose of writing, and then the reaction to a read-only PV makes sense.

How about keeping it simple and displaying the PV in question by other means, for example as a label or LED right next to the button?

gilesknap, 2015-07-21:
We are trying to repeat the functionality that we have on our beamline's EDM screens. A symbol widget represents a component such as a set of slits. Clicking on it opens another opi which shows all of the individual slit devices (motors, temperatures etc.). We then have a hierarchical set of status PVs which all propagate their alarm state up the pv tree. The root status PV is attached to the symbol widget and thus if any of the devices are in alarm state the symbol widget shows the alarm.

A second widget beside the symbol to indicate error state would be a shame since we are trying to create a very compact representation of the beamline so that it can be always shown and used for monitoring as well as the entry point to the rest of the opis.

Since it is possible to have many different default actions for an action button it is a pity to limit those that do not write to the pv in the readonly case.

This e-mail and any attachments may contain confidential, copyright and or privileged material, and are for the use of the intended addressee only. If you are not the intended addressee or an authorised recipient of the addressee please notify us of receipt by returning the e-mail and do not use, copy, retain, distribute or disclose the information in or attached to the e-mail.
Any opinions expressed within this e-mail are those of the individual and not necessarily of Diamond Light Source Ltd. 
Diamond Light Source Ltd. cannot guarantee that this e-mail or any attachments are free from viruses and we cannot accept liability for any damage which you may sustain as a result of software viruses which may be transmitted in or with the message.
Diamond Light Source Limited (company no. 4375679). Registered in England and Wales with its registered office at Diamond House, Harwell Science and Innovation Campus, Didcot, Oxfordshire, OX11 0DE, United Kingdom

kasemir, 2015-07-21:
My concern is to keep BOY general. Text updates, LEDs, .. display a PV, action buttons write a PV.
If the PV cannot be written, the action button becomes disabled.

Turning the action button for opening screen X into an alarm summary indicator is a special case. I don't want to add a new "Disable if PV read-only" option to the button just because one site needed this, if we can find a more generic solution.

Can you try using two widgets on top of each other? Something with alarm sensitive border behind the button? That way, it's a plain alarm-sensitive-border widget and a plain open-display button, nothing special about either one of them.

How many of these are there? Would it be an option to use a rule for the border color of the button?

Since you mention that this is for beam lines: Especially on beam lines, where you have users who only visit for a short time, I think it's a good idea to keep it simple and straight forward. Labels, LEDs, text updates are easily understood. A video-game screen where every colored border has a special meaning requires a learning period that users who only visit the beam line for a short time don't have. Plus next you get into trouble with color-blindness, so a label "XYZ not OK" is needed next to each colored border anyway.

kasemir, 2015-07-29:
We just had a somewhat related case, an ActionButton to "stop" several motors. The button had multiple "Write PV" actions, i.e. one per motor, and "Hook all actions to mouse click" was selected.
So pressing the button would perform all actions and thus stop all motors.
.. except when one PV was disconnected, the complete button becomes disabled, you can't stop any motor.

Our conclusion was to not implement automation (doing a sequence of writes) in the GUI, and instead write to just one PV, which is a record on some auxiliary IOC which has dfanout records to stop all the motors.

willrogers, 2015-08-07:
After some discussion, we are going to accept this decision and work around our problem.

closed 2015-08-07 by willrogers