ControlSystemStudio/cs-studio #1264: databrowser: archive appliance time issue

berryma4, 2015-07-21:
the data from the archive appliance:
[image]

the data as databrowser sees it:
[image]

Notice the year is incorrect starting at the highlighted line in databrowser.

berryma4, 2015-07-22:
@kasemir, I haven't looked into this.  Does this look like a databrowser or appliance plugin issue.

kasemir, 2015-07-22:
Your data crosses the daylight saving time change EST/EDT.
The data browser doesn't really care about this because all the data that it sees is VType samples with a timestamp, and that's in seconds relative to 1/1/1970 UTC.

When the data browser displays times on the time axis or in the sample view, that's always converted to local time. So when time jumps from normal time to DST on the morning of March 8, it just jumps from 01:59 to 03:00 in the display.

Your data has "Dec/13/2014 00:27:27 EST" converted into "2014/12/13 00:27:26", so that looks OK.
But then the next sample from the archive appliance, "Jul/15/2015 14:29:25 EDT" seems to get parsed into a VType sample with time stamp 2014/07/15 14:29:25. The time and also the value look OK, but the date is a year off. Jul/15/2015  -> 2014/07/15.

My guess would be an error in org.csstudio.archive.reader.appliance. I tried to look at that, but it calls into some pbrawclient.jar binary, so no idea how it receives and maybe parses the time stamps.

[image]

berryma4, 2015-07-27:
@jbobnar @shroffk do you see this?

berryma4, 2015-08-10:
pbrawclient update fixed issue

closed 2015-08-10 by berryma4