ControlSystemStudio/cs-studio #1270: databrowser: archive appliance falls back to raw data

berryma4, 2015-07-27:
For some reason, databrowser keeps falling back to raw data (not optimized), which cs-studio kills because that is way too many points (see attached dump). @kasemir @jbobnar 
[image]

jbobnar, 2015-07-27:
I just tried this locally and it works fine.

Which version of appliance are you using? Do you have a chance to step through with a debugger? If yes, put a breakpoint on line 142 in ApplianceArchiveReader and let us know what happens there.

berryma4, 2015-07-27:
I'm updating to pbrawclient 0.0.4, it seems to fix the issue.

berryma4, 2015-07-27:
This did not completely fix the issue

berryma4, 2015-07-27:
strange with this in my preferences file:
org.csstudio.archive.reader.appliance/useStatisticsForOptimizedData=true
ApplianceArchiveReader in line 148, useStatistics is sometimes false?

Now they are working?
But, the stats look wrong:
[image]

berryma4, 2015-07-27:
the problem may be with the archive appliance ... (other than the useStatics sometimes false).
When ApplianceArchiveReader:142 sometimes comes in as zero, then fails into the catch which is a raw value iterator.

I'm not sure if I like this failing over to raw.  This will kill most clients pretty fast.

berryma4, 2015-07-27:
After removing from the preferences, I can't seem to reproduce this.
Although the stats are strange:
[image]

berryma4, 2015-07-27:
All I can think was maybe the end of line in the preferences file was messed up?

berryma4, 2015-07-27:
End of lines were the same, I'm not sure what's going on here.
[image]

berryma4, 2015-07-27:
Also, ApplianceArchiveReader ln:142 sometimes returns 0
then uses the raw iterator.
I'm guessing this is a appliance issue?

berryma4, 2015-07-27:
Yes, this is the main issue ... ncount returns zero sometimes.

kasemir, 2015-07-27:
That logic in ApplianceArchiveReader:142 seems odd.

The optimization
```
   ncount = getNumberOfSamples()
   if (ncount is small)
       getRawData()
   else
       getOptimizedData()
```
should be on the server side, i.e. the archive appliance side.
If the client asks first for the count, then for the data, it's an extra network trip that could be avoided. The client should just ask for the optimized data. If the server only has 10 samples for the requested time range, it returns those raw samples.

jbobnar, 2015-07-28:
Extra network trip is cheap. Counting the points (which the server needs to do regardless of if it is done in a separate request or not) is much more expensive. That is specially true for waveforms. I'll have a look at the appliance code if there is something easy that we can do.

berryma4, 2015-07-28:
Thank you, fixing the zero ncount would help a lot.

jbobnar, 2015-07-30:
@berryma4 Does ncount=0 happens every time when you make the request with specific start/end time parameters and specific PV, or is it random (using the same parameters and the same PV ncount sometimes returns 0, the other times it returns a proper value)?

If it happens every time when you use a specific time window it could be that one of the samples has a messed up timestamp (somewhere in the future). Appliance expects that samples are stored in timestamp, so all further samples (which have timestamp smaller than the problematic one) are skipped. In such case, other processors such as mean and std would also give you some strange results. If the problem is random, than it is an issue in the appliance.

I wrote a new post processor to retrieve all statistical data in one call, which will speed up the data loading process. I'll ask Murali to add it, but it won't solve the ncount issue...

berryma4, 2015-08-10:
Still waiting on the changes to make it in to the archive appliance.  But, we can close this issue.  Thank you!

closed 2015-08-10 by berryma4