ControlSystemStudio/cs-studio #1274: databrowser error

berryma4, 2015-07-29:
@kasemir have you seen this?
```java
2015-07-29 10:25:43.434 WARNING [Thread 46] org.epics.pvmanager.util.Executors$2 (execute) - Exception on the timer thread caused by a ValueListener
java.lang.RuntimeException: Cannot locate trace for L098RF.PHASE3
	at org.csstudio.trends.databrowser2.ui.ModelBasedPlot.findTrace(ModelBasedPlot.java:326)
	at org.csstudio.trends.databrowser2.ui.ModelBasedPlot.updateTrace(ModelBasedPlot.java:298)
	at org.csstudio.trends.databrowser2.ui.Controller$4.changedItemLook(Controller.java:486)
	at org.csstudio.trends.databrowser2.model.Model.fireItemLookChanged(Model.java:936)
	at org.csstudio.trends.databrowser2.model.ModelItem.fireItemLookChanged(ModelItem.java:198)
	at org.csstudio.trends.databrowser2.model.ModelItem.setUnits(ModelItem.java:169)
	at org.csstudio.trends.databrowser2.model.PVItem.updateUnits(PVItem.java:393)
	at org.csstudio.trends.databrowser2.model.PVItem.pvChanged(PVItem.java:380)
	at org.epics.pvmanager.PVReaderImpl.firePvValueChanged(PVReaderImpl.java:110)
	at org.epics.pvmanager.PVReaderImpl.setValue(PVReaderImpl.java:220)
	at org.epics.pvmanager.PVDirector$4.run(PVDirector.java:343)
	at org.epics.pvmanager.util.Executors$2.execute(Executors.java:55)
	at org.epics.pvmanager.PVDirector.notifyPv(PVDirector.java:314)
	at org.epics.pvmanager.PVDirector.access$500(PVDirector.java:37)
	at org.epics.pvmanager.PVDirector$6.desiredRateEvent(PVDirector.java:388)
	at org.epics.pvmanager.SourceDesiredRateDecoupler.sendDesiredRateEvent(SourceDesiredRateDecoupler.java:169)
	at org.epics.pvmanager.PassiveScanDecoupler$1.run(PassiveScanDecoupler.java:50)
	at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(Unknown Source)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
```

kasemir, 2015-07-29:
I do see some "Cannot locate trace for .." messages in the CSS logs on the accelerator.
I'm not sure how to reproduce.

berryma4, 2015-07-29:
I also get this one:
```java 
2015-07-29 16:21:42.616 WARNING [Thread 49] org.epics.pvmanager.util.Executors$2 (execute) - Exception on the timer thread caused by a ValueListener
org.eclipse.swt.SWTException: Invalid thread access
	at org.eclipse.swt.SWT.error(SWT.java:4441)
	at org.eclipse.swt.SWT.error(SWT.java:4356)
	at org.eclipse.swt.SWT.error(SWT.java:4327)
	at org.eclipse.swt.widgets.Widget.error(Widget.java:476)
	at org.eclipse.swt.widgets.Widget.checkWidget(Widget.java:367)
	at org.eclipse.swt.widgets.Table.deselectAll(Table.java:1980)
	at org.eclipse.jface.viewers.TableViewer.doDeselectAll(TableViewer.java:300)
	at org.eclipse.jface.viewers.AbstractTableViewer.setSelectionToWidget(AbstractTableViewer.java:838)
	at org.eclipse.jface.viewers.StructuredViewer.setSelectionToWidget(StructuredViewer.java:1751)
	at org.eclipse.jface.viewers.StructuredViewer.preservingSelection(StructuredViewer.java:1443)
	at org.eclipse.jface.viewers.StructuredViewer.preservingSelection(StructuredViewer.java:1397)
	at org.eclipse.jface.viewers.StructuredViewer.refresh(StructuredViewer.java:1497)
	at org.eclipse.jface.viewers.ColumnViewer.refresh(ColumnViewer.java:521)
	at org.csstudio.trends.databrowser2.propsheet.TraceTableHandler$1.changedItemLook(TraceTableHandler.java:108)
	at org.csstudio.trends.databrowser2.model.Model.fireItemLookChanged(Model.java:936)
	at org.csstudio.trends.databrowser2.model.ModelItem.fireItemLookChanged(ModelItem.java:198)
	at org.csstudio.trends.databrowser2.model.ModelItem.setUnits(ModelItem.java:169)
	at org.csstudio.trends.databrowser2.model.PVItem.updateUnits(PVItem.java:393)
	at org.csstudio.trends.databrowser2.model.PVItem.pvChanged(PVItem.java:380)
	at org.epics.pvmanager.PVReaderImpl.firePvValueChanged(PVReaderImpl.java:110)
	at org.epics.pvmanager.PVReaderImpl.setValue(PVReaderImpl.java:220)
	at org.epics.pvmanager.PVDirector$4.run(PVDirector.java:343)
	at org.epics.pvmanager.util.Executors$2.execute(Executors.java:55)
	at org.epics.pvmanager.PVDirector.notifyPv(PVDirector.java:314)
	at org.epics.pvmanager.PVDirector.access$500(PVDirector.java:37)
	at org.epics.pvmanager.PVDirector$6.desiredRateEvent(PVDirector.java:388)
	at org.epics.pvmanager.SourceDesiredRateDecoupler.sendDesiredRateEvent(SourceDesiredRateDecoupler.java:169)
	at org.epics.pvmanager.PassiveScanDecoupler$1.run(PassiveScanDecoupler.java:50)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
```

berryma4, 2015-07-29:
Here is my problem ....
When I run from the IDE I get this:
[image]

While my tycho built I get this:
[image]

berryma4, 2015-07-29:
@shroffk or @jbobnar you use the archive appliance, .... do you see this issue?

kasemir, 2015-07-29:
Found the reason for the invalid thread access.

kasemir, 2015-07-29:
In those cases where you see fewer samples, is that the same time when you see the error messages? 
On the  issue_1274 branch, the archive access will log (in FINE mode) when archive access started and stopped, with number of samples received. Maybe that will show if the difference is in the data you get, or in the way its displayed.

jbobnar, 2015-07-30:
I saw the Invalid Thread Access a few times, but not the other problem. 
Check that you use the same java version and that tycho uses all the latest & greatest bundles. It drove me bananas a few times when the code was changed, but the version number remained the same.

berryma4, 2015-07-30:
@kasemir with the issue_1274 branch I get this error.
(when tycho compiled, but not in the IDE run) 
```java
2015-07-30 13:33:34.717 FINE [Thread 158] org.csstudio.trends.databrowser2.ui.Controller$1 (archiveFetchFailed) - No archived data for AMPL RD
org.csstudio.archive.reader.UnknownChannelException: Channel not found in archive: L082RF.AMPL3
	at org.csstudio.archive.reader.appliance.ApplianceArchiveReader.getRawValues(ApplianceArchiveReader.java:131)
	at org.csstudio.archive.reader.appliance.ApplianceArchiveReader.getRawValues(ApplianceArchiveReader.java:1)
	at org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread.run(ArchiveFetchJob.java:123)
	at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

2015-07-30 13:33:34.717 FINE [Thread 158] org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread (run) - Ended Read data: L082RF.AMPL3, 2015-07-30 10:53:28.354 -0400 - 2015-07-30 13:33:34.097 -0400 with 0 samples in 0.615 seconds
2015-07-30 13:33:34.717 FINE [Thread 153] org.csstudio.trends.databrowser2.ui.Controller$1 (archiveFetchFailed) - No archived data for L082RF.STEPRPOS
org.csstudio.archive.reader.UnknownChannelException: Channel not found in archive: L082RF.STEPRPOS
	at org.csstudio.archive.reader.appliance.ApplianceArchiveReader.getRawValues(ApplianceArchiveReader.java:131)
	at org.csstudio.archive.reader.appliance.ApplianceArchiveReader.getRawValues(ApplianceArchiveReader.java:1)
	at org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread.run(ArchiveFetchJob.java:123)
	at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

2015-07-30 13:33:34.717 FINE [Thread 153] org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread (run) - Ended Read data: L082RF.STEPRPOS, 2015-07-30 10:53:28.354 -0400 - 2015-07-30 13:33:34.097 -0400 with 0 samples in 0.615 seconds
2015-07-30 13:33:34.753 FINE [Thread 51] org.csstudio.trends.databrowser2.model.PVItem (pvChanged) - PV L084RF.AMPL1 received VFloat[206.38474, 2015/07/30 13:33:34.806]
2015-07-30 13:33:34.753 FINE [Thread 50] org.csstudio.trends.databrowser2.model.PVItem (pvChanged) - PV L085RF.AMPL1 received VFloat[229.04826, 2015/07/30 13:33:34.806]
2015-07-30 13:33:34.808 FINE [Thread 48] org.csstudio.trends.databrowser2.model.PVItem (pvChanged) - PV L091RF.AMPL1 received VFloat[111.82707, 2015/07/30 13:33:34.861]
2015-07-30 13:33:34.851 FINE [Thread 156] org.csstudio.trends.databrowser2.ui.Controller$1 (archiveFetchFailed) - No archived data for L082RF.PIEZOPOS
org.csstudio.archive.reader.UnknownChannelException: Channel not found in archive: L082RF.PIEZOPOS
	at org.csstudio.archive.reader.appliance.ApplianceArchiveReader.getRawValues(ApplianceArchiveReader.java:131)
	at org.csstudio.archive.reader.appliance.ApplianceArchiveReader.getRawValues(ApplianceArchiveReader.java:1)
	at org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread.run(ArchiveFetchJob.java:123)
	at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

2015-07-30 13:33:34.851 FINE [Thread 156] org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread (run) - Ended Read data: L082RF.PIEZOPOS, 2015-07-30 10:53:28.354 -0400 - 2015-07-30 13:33:34.097 -0400 with 0 samples in 0.748 seconds
2015-07-30 13:33:34.854 FINE [Thread 157] org.csstudio.trends.databrowser2.ui.Controller$1 (archiveFetchFailed) - No archived data for FW power
org.csstudio.archive.reader.UnknownChannelException: Channel not found in archive: L082RF.AMPL1
	at org.csstudio.archive.reader.appliance.ApplianceArchiveReader.getRawValues(ApplianceArchiveReader.java:131)
	at org.csstudio.archive.reader.appliance.ApplianceArchiveReader.getRawValues(ApplianceArchiveReader.java:1)
	at org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread.run(ArchiveFetchJob.java:123)
	at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
```

berryma4, 2015-07-30:
And I'm still getting this error:
```java
2015-07-30 13:33:33.701 WARNING [Thread 51] org.epics.pvmanager.util.Executors$2 (execute) - Exception on the timer thread caused by a ValueListener
java.lang.RuntimeException: Cannot locate trace for L088RF.AMPL1
	at org.csstudio.trends.databrowser2.ui.ModelBasedPlot.findTrace(ModelBasedPlot.java:326)
	at org.csstudio.trends.databrowser2.ui.ModelBasedPlot.updateTrace(ModelBasedPlot.java:298)
	at org.csstudio.trends.databrowser2.ui.Controller$4.changedItemLook(Controller.java:486)
	at org.csstudio.trends.databrowser2.model.Model.fireItemLookChanged(Model.java:936)
	at org.csstudio.trends.databrowser2.model.ModelItem.fireItemLookChanged(ModelItem.java:198)
	at org.csstudio.trends.databrowser2.model.ModelItem.setUnits(ModelItem.java:169)
	at org.csstudio.trends.databrowser2.model.PVItem.updateUnits(PVItem.java:393)
	at org.csstudio.trends.databrowser2.model.PVItem.pvChanged(PVItem.java:380)
	at org.epics.pvmanager.PVReaderImpl.firePvValueChanged(PVReaderImpl.java:110)
	at org.epics.pvmanager.PVReaderImpl.setValue(PVReaderImpl.java:220)
	at org.epics.pvmanager.PVDirector$4.run(PVDirector.java:343)
	at org.epics.pvmanager.util.Executors$2.execute(Executors.java:55)
	at org.epics.pvmanager.PVDirector.notifyPv(PVDirector.java:314)
	at org.epics.pvmanager.PVDirector.access$500(PVDirector.java:37)
	at org.epics.pvmanager.PVDirector$6.desiredRateEvent(PVDirector.java:388)
	at org.epics.pvmanager.SourceDesiredRateDecoupler.sendDesiredRateEvent(SourceDesiredRateDecoupler.java:169)
	at org.epics.pvmanager.PassiveScanDecoupler$1.run(PassiveScanDecoupler.java:50)
	at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(Unknown Source)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

2015-07-30 13:33:33.702 WARNING [Thread 50] org.epics.pvmanager.util.Executors$2 (execute) - Exception on the timer thread caused by a ValueListener
java.lang.RuntimeException: Cannot locate trace for L091RF.AMPL1
	at org.csstudio.trends.databrowser2.ui.ModelBasedPlot.findTrace(ModelBasedPlot.java:326)
	at org.csstudio.trends.databrowser2.ui.ModelBasedPlot.updateTrace(ModelBasedPlot.java:298)
	at org.csstudio.trends.databrowser2.ui.Controller$4.changedItemLook(Controller.java:486)
	at org.csstudio.trends.databrowser2.model.Model.fireItemLookChanged(Model.java:936)
	at org.csstudio.trends.databrowser2.model.ModelItem.fireItemLookChanged(ModelItem.java:198)
	at org.csstudio.trends.databrowser2.model.ModelItem.setUnits(ModelItem.java:169)
	at org.csstudio.trends.databrowser2.model.PVItem.updateUnits(PVItem.java:393)
	at org.csstudio.trends.databrowser2.model.PVItem.pvChanged(PVItem.java:380)
	at org.epics.pvmanager.PVReaderImpl.firePvValueChanged(PVReaderImpl.java:110)
	at org.epics.pvmanager.PVReaderImpl.setValue(PVReaderImpl.java:220)
	at org.epics.pvmanager.PVDirector$4.run(PVDirector.java:343)
	at org.epics.pvmanager.util.Executors$2.execute(Executors.java:55)
	at org.epics.pvmanager.PVDirector.notifyPv(PVDirector.java:314)
	at org.epics.pvmanager.PVDirector.access$500(PVDirector.java:37)
	at org.epics.pvmanager.PVDirector$6.desiredRateEvent(PVDirector.java:388)
	at org.epics.pvmanager.SourceDesiredRateDecoupler.sendDesiredRateEvent(SourceDesiredRateDecoupler.java:169)
	at org.epics.pvmanager.PassiveScanDecoupler$1.run(PassiveScanDecoupler.java:50)
	at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(Unknown Source)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
```

berryma4, 2015-07-30:
Of note, re opening the plot it found L082RF.STEPRPOS
```java
2015-07-30 13:38:25.740 FINE [Thread 206] org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread (run) - Ended Read data: L082RF.STEPRPOS, 2015-07-30 10:46:23.555 -0400 - 2015-07-30 13:26:35.298 -0400 with 3 samples in 0.528 seconds
```

even though it got this previously (posted above https://github.com/ControlSystemStudio/cs-studio/issues/1274#issuecomment-126412655):
```java
org.csstudio.archive.reader.UnknownChannelException: Channel not found in archive: L082RF.STEPRPOS
```

kasemir, 2015-07-30:
In our setup, we list several archive sources: One for instuments, one for accelerator. When looking for data, at most one source contains the channel. So data is fetched from it, while the others report that error, which is only logged at the FINE level because it's nothing the end user needs to worry about.

Do you get the error when using only one data source?

On the  "Cannot locate trace", I don't know how to reproduce it.

berryma4, 2015-07-30:
Yes I have only one archive datasource.
In the IDE I get everything working fine:
```java
2015-07-30 14:08:26.165 INFO [Thread 142] org.epics.archiverappliance.retrieval.client.RawDataRetrieval (getDataForPV) - URL to fetch data is http://epicsarchiver.nscl.msu.edu/retrieval/data/getData.raw?pv=ncount%28L084RF.AMPL1%29&from=2015-07-30T11%3A28%3A20.421-04%3A00&to=2015-07-30T14%3A08%3A26.164-04%3A00
2015-07-30 14:08:26.167 FINE [Thread 144] org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread (run) - Starting Read data: L085RF.AMPL1, 2015-07-30 11:28:20.421 -0400 - 2015-07-30 14:08:26.164 -0400
2015-07-30 14:08:26.167 INFO [Thread 144] org.epics.archiverappliance.retrieval.client.RawDataRetrieval (getDataForPV) - URL to fetch data is http://epicsarchiver.nscl.msu.edu/retrieval/data/getData.raw?pv=ncount%28L085RF.AMPL1%29&from=2015-07-30T11%3A28%3A20.421-04%3A00&to=2015-07-30T14%3A08%3A26.164-04%3A00
2015-07-30 14:08:26.168 FINE [Thread 147] org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread (run) - Starting Read data: L082RF.AMPL1, 2015-07-30 11:28:20.421 -0400 - 2015-07-30 14:08:26.164 -0400
2015-07-30 14:08:26.168 INFO [Thread 147] org.epics.archiverappliance.retrieval.client.RawDataRetrieval (getDataForPV) - URL to fetch data is http://epicsarchiver.nscl.msu.edu/retrieval/data/getData.raw?pv=L082RF.AMPL1&from=2015-07-30T11%3A28%3A20.421-04%3A00&to=2015-07-30T14%3A08%3A26.164-04%3A00
2015-07-30 14:08:26.169 FINE [Thread 145] org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread (run) - Starting Read data: L082RF.PIEZOPOS, 2015-07-30 11:28:20.421 -0400 - 2015-07-30 14:08:26.164 -0400
2015-07-30 14:08:26.169 INFO [Thread 145] org.epics.archiverappliance.retrieval.client.RawDataRetrieval (getDataForPV) - URL to fetch data is http://epicsarchiver.nscl.msu.edu/retrieval/data/getData.raw?pv=L082RF.PIEZOPOS&from=2015-07-30T11%3A28%3A20.421-04%3A00&to=2015-07-30T14%3A08%3A26.164-04%3A00
2015-07-30 14:08:26.169 FINE [Thread 149] org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread (run) - Starting Read data: L081HP, 2015-07-30 11:28:20.421 -0400 - 2015-07-30 14:08:26.164 -0400
2015-07-30 14:08:26.169 INFO [Thread 149] org.epics.archiverappliance.retrieval.client.RawDataRetrieval (getDataForPV) - URL to fetch data is http://epicsarchiver.nscl.msu.edu/retrieval/data/getData.raw?pv=ncount%28L081HP%29&from=2015-07-30T11%3A28%3A20.421-04%3A00&to=2015-07-30T14%3A08%3A26.164-04%3A00
2015-07-30 14:08:26.170 FINE [Thread 150] org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread (run) - Starting Read data: L082RF.AMPL3, 2015-07-30 11:28:20.421 -0400 - 2015-07-30 14:08:26.164 -0400
2015-07-30 14:08:26.170 INFO [Thread 150] org.epics.archiverappliance.retrieval.client.RawDataRetrieval (getDataForPV) - URL to fetch data is http://epicsarchiver.nscl.msu.edu/retrieval/data/getData.raw?pv=L082RF.AMPL3&from=2015-07-30T11%3A28%3A20.421-04%3A00&to=2015-07-30T14%3A08%3A26.164-04%3A00
2015-07-30 14:08:26.170 FINE [Thread 146] org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread (run) - Starting Read data: L088RF.AMPL1, 2015-07-30 11:28:20.421 -0400 - 2015-07-30 14:08:26.164 -0400
2015-07-30 14:08:26.170 INFO [Thread 146] org.epics.archiverappliance.retrieval.client.RawDataRetrieval (getDataForPV) - URL to fetch data is http://epicsarchiver.nscl.msu.edu/retrieval/data/getData.raw?pv=ncount%28L088RF.AMPL1%29&from=2015-07-30T11%3A28%3A20.421-04%3A00&to=2015-07-30T14%3A08%3A26.164-04%3A00
2015-07-30 14:08:26.171 FINE [Thread 152] org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread (run) - Starting Read data: L082RF.AMPL2, 2015-07-30 11:28:20.421 -0400 - 2015-07-30 14:08:26.164 -0400
2015-07-30 14:08:26.171 INFO [Thread 152] org.epics.archiverappliance.retrieval.client.RawDataRetrieval (getDataForPV) - URL to fetch data is http://epicsarchiver.nscl.msu.edu/retrieval/data/getData.raw?pv=L082RF.AMPL2&from=2015-07-30T11%3A28%3A20.421-04%3A00&to=2015-07-30T14%3A08%3A26.164-04%3A00
2015-07-30 14:08:26.171 FINE [Thread 143] org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread (run) - Starting Read data: L091RF.AMPL1, 2015-07-30 11:28:20.421 -0400 - 2015-07-30 14:08:26.164 -0400
2015-07-30 14:08:26.171 INFO [Thread 143] org.epics.archiverappliance.retrieval.client.RawDataRetrieval (getDataForPV) - URL to fetch data is http://epicsarchiver.nscl.msu.edu/retrieval/data/getData.raw?pv=ncount%28L091RF.AMPL1%29&from=2015-07-30T11%3A28%3A20.421-04%3A00&to=2015-07-30T14%3A08%3A26.164-04%3A00
2015-07-30 14:08:26.172 FINE [Thread 148] org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread (run) - Starting Read data: L082RF.STEPRPOS, 2015-07-30 11:28:20.421 -0400 - 2015-07-30 14:08:26.164 -0400
2015-07-30 14:08:26.172 INFO [Thread 148] org.epics.archiverappliance.retrieval.client.RawDataRetrieval (getDataForPV) - URL to fetch data is http://epicsarchiver.nscl.msu.edu/retrieval/data/getData.raw?pv=L082RF.STEPRPOS&from=2015-07-30T11%3A28%3A20.421-04%3A00&to=2015-07-30T14%3A08%3A26.164-04%3A00
2015-07-30 14:08:26.172 FINE [Thread 151] org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread (run) - Starting Read data: L082RF_AMPL, 2015-07-30 11:28:20.421 -0400 - 2015-07-30 14:08:26.164 -0400
2015-07-30 14:08:26.172 INFO [Thread 151] org.epics.archiverappliance.retrieval.client.RawDataRetrieval (getDataForPV) - URL to fetch data is http://epicsarchiver.nscl.msu.edu/retrieval/data/getData.ra [... cut ...]
2015-07-30 14:08:28.512 FINE [Thread 43] org.csstudio.trends.databrowser2.model.PVItem (pvChanged) - PV G166PG received VDouble[0.003411166137084365, 2015/07/30 14:08:28.524]
2015-07-30 14:08:28.528 FINE [Thread 143] org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread (run) - Ended Read data: L091RF.AMPL1, 2015-07-30 11:28:20.421 -0400 - 2015-07-30 14:08:26.164 -0400 with 802 samples in 2.356 seconds
2015-07-30 14:08:28.593 FINE [Thread 144] org.csstudio.trends.databrowser2.archive.ArchiveFetchJob$WorkerThread (run) - Ended Read data: L085RF.AMPL1, 2015-07-30 11:28:20.421 -0400 - 2015-07-30 14:08:26.164 -0400 with 802 samples in 2.426 seconds
```

berryma4, 2015-07-30:
This is where I get the throw if I attach a debugger to a tycho built product.

```java
	protected void fetchDataInternal(String pvName) throws ArchiverApplianceException {				
		java.sql.Timestamp sqlStartTimestamp = TimestampHelper.toSQLTimestamp(start);
		java.sql.Timestamp sqlEndTimestamp = TimestampHelper.toSQLTimestamp(end);
		
		DataRetrieval dataRetrieval = reader.createDataRetriveal(reader.getDataRetrievalURL());
		mainStream = dataRetrieval.getDataForPV(pvName, sqlStartTimestamp, sqlEndTimestamp);
		if (mainStream != null) { 
			mainIterator = mainStream.iterator();
		} else {
			throw new ArchiverApplianceException("Could not fetch data.");
		}
```

Which is strange, the appliance is answering fine.  I did a 
```unix
watch -n 1 wget -O- url
```
I let it run for the afternoon, it never got anything but 200.

Is there something different I'm missing between the IDE and tycho build that could cause this?

berryma4, 2015-07-30:
ahhhhh, these preferences actually apply to the archive appliance. 
```unix
org.eclipse.core.net/proxyData/HTTP/host=
org.eclipse.core.net/proxyData/HTTP/port=
org.eclipse.core.net/proxyData/HTTP/hasAuth=
org.eclipse.core.net/proxyData/HTTPS/host=
org.eclipse.core.net/proxyData/HTTPS/port=
org.eclipse.core.net/proxyData/HTTPS/hasAuth=
```
I feel dumb, and that I should have checked wireshark at the beginning.  I guess the IDE run must ignore that preference setting.

berryma4, 2015-07-30:
Is there a way to add the archive appliance as an exception?

jbobnar, 2015-07-31:
This might work:
```
org.eclipse.core.net/nonProxiedHosts=epicsarchiver.nscl.msu.edu
```
You can also edit the settings in Preferences -> General -> Network Connections

berryma4, 2015-07-31:
I'm going to push the Issue_1274 branch on master and 4.1.x, for the thread access fix.

As for the webproxy, everything works fine if the archiver hostname is excluded.
Strange is, looking at wireshark, I get 200s for everything though the proxy with all the data.  I guess I need to look at the headers in more detail and attach the appliance protobuf source code.

berryma4, 2015-08-10:
closing issue, .. problem was caused by a web proxy update.
Although, I the new log and the "Invalid thread access" fix were a nice addition.  Thank you!

closed 2015-08-10 by berryma4