ControlSystemStudio/cs-studio #1275: Ability to run script when an OPI is closed down

dmichel76, 2015-07-29:
If scripts can be ran when an OPI starts up using the "=1" trigger PV, it would be very useful to be able to run script when an OPI is closed down. This could be done using a similar 'trick' e.g. a "=0" trigger PV

kasemir, 2015-07-29:
What is an example use case?

The use of scripts should be very restricted, only about display matters. They're not meant to perform automation or the like, that has to be on an IOC.
What would a close-down script do? Can't perform any display function, because the display is being closed. I can only come up with really bad examples, like starting a timer or thread at startup, which you then want to stop on close.

Those who insist to shoot their foot with scripts that do more than they should can use
```
widget.getWidgetModel().getRootDisplayModel().getViewer().getControl().addDisposeListener(
  YourDisposeListener()
)
```
to perform actions when the display closes.
But that should really be reserved for people who don't have to ask about this, who are willing to adapt their scripts with every update of CSS.

dmichel76, 2015-07-30:
I agree, scripting should only be about the user interface! Without a very specific use case in mind, I was thinking about whether an OPI could notify another OPI that it's been closed, so, for instance, the main navigation OPI could then change according to what is currently open, and what is not.

But I understand that it probably gives away too much power and makes very tempting to "shoot their foot with scripts".

berryma4, 2015-09-29:
ok to close?

closed 2018-06-12 by berryma4