ControlSystemStudio/cs-studio #1276: Ability to specify the x,y position of an OPI window

dmichel76, 2015-07-29:
It would be useful to be able to specify the position on screen of an OPI window. This would give the developer a lot more control over the overall user-experience, especially when multiple monitors are used (as it is often the case in control rooms)

Several features could be considered for this:
- specify x and y and monitor number
- specify an offset from the parent window

willrogers, 2015-08-03:
Hi David,

Are you keeping up with the changes on master branch?  There are now light-weight OPI windows, which don't include all the menus and trimmings that a regular CSS window has.  If I remember CLF's requirements correctly, that might be useful to you.

The OPI windows honour the X and Y position specified in the XML, but how would you like this to be configurable?  If you're opening full Eclipse windows, how are you doing it and how would you like to configure their locations?

Cheers,
Will

berryma4, 2015-09-29:
@dmichel76 ok to close?

closed 2018-06-12 by berryma4