ControlSystemStudio/cs-studio #1278: Ability to NOT remember the last opened OPI

dmichel76, 2015-07-29:
This feature, which can be very useful, can also create problems. With multiple windows opened, a user might end up closing windows down in a random order. Opening CSS again will open the last opened window which may not be appropriate - it might be more appropriate to open the main navigation screen or synoptic, for example.

Opening a specific OPI at startup is easily done using the --launcher.openFile option on the command line, so there is no need to implement a new feature for that. However, with the "remember the last opened OPI window" feature active, this will result in having 2 OPIs opened: the one specified in the command line, but also the last window opened, making it difficult to setup a clean startup look for client machines.

kasemir, 2015-07-29:
This is more of an Eclipse feature that's not easy to change.

.. but it's not too hard to live with it.
Your 'main navigation screen' should be available from the top-opis, so it's always easy to restore the initial state: Close all, then open your main navigation screen.

With the recent work on using views to execute the displays, you can also save your layouts in perspectives. So to get a "read only" setup that starts over as you want, start CSS with a script that restores your golden workbench.xmi
--> Your users get the default setup, and can then switch between perspectives.

dmichel76, 2015-07-30:
"Eclipse feature that's not easy to change"... That's what I thought you guys might say.

It's good to know about the ability to save layouts as a perspective! When you say "recent work", does that mean it's still in development or is already released? and if so, is there anything online that explains how to do this?

kasemir, 2015-07-30:
The master branch has been updated to execute *.opi displays in "views".
See https://github.com/ControlSystemStudio/cs-studio/issues/544, especially the June 1 entry on what's new.

berryma4, 2015-09-29:
@dmichel76 ok to close?